motioneye-project/motioneye #780: How to start the camera interface from Raspbian, where it is installed.

Gosta49, 2018-01-30:
I have installed motioneye on Raspbian (Pi 3).
Is it possible to start the camera interface on Raspbian without going through the browser(localhost)/server?
There is probably an answer somewhere but I haven't found it.

ccrisan, 2018-01-31:
What do you mean by "start the camera interface"?

Gosta49, 2018-01-31:
Hi ccrisan,
I want to see my camera images on the Raspberry Pi where the camera and motioneye are installed. I hope there is another way than using the browser ..?

ccrisan, 2018-02-01:
You could enable *Fast Network Camera* and *HDMI Preview*. Keep in mind though that you'll lose all "intelligent" functions such as motion detection, movie creation, notifications, etc.

Gosta49, 2018-02-02:
Thank you for answering ccrisan,
but I can't figure out how to enable 'Fast Network Camera' and 'HDMI Preview' to see the USB camera image on Rasbian. Is there a GUI (python file) to start or..? (I still have a lot to learn about Linux).
'motion' is running on Raspbian and the USB camera (video0) is then not accessible from other programs like 'fswebcam'.
It seems that my question is a bit odd, but I have my reasons ...

ccrisan, 2018-02-02:
@Gosta49 unfortunately Fast Network Camera + HDMI Preview only works for the PRi CSI camera. Local preview for USB or netcams is not possible with motionEye(OS).

closed 2018-02-02 by ccrisan