motioneye-project/motioneye #781: OSError: [Errno 2] No such file or directory (when starting with systemd)

jwc866, 2018-01-31:
Hello:
I recently installed motioneye on a new CentOS7 system.  I have followed the Fedora instructions to ensure all required packages are present.  I did not install motion, ffmpeg and v4l-utils as I am only using motioneye as a hub.  When starting the motioneye service with systemd, the service starts and stops right away.  Systemd shows the below message.  I am not sure which file it is looking for, can you help?  Thank you for your help again.

Output from systemd:

● motioneye.service - motionEye Server
   Loaded: loaded (/etc/systemd/system/motioneye.service; enabled; vendor preset: disabled)
   Active: failed (Result: exit-code) since Wed 2018-01-31 02:20:46 UTC; 3s ago
  Process: 1613 ExecStart=/usr/bin/meyectl startserver -c /etc/motioneye/motioneye.conf (code=exited, status=1/FAILURE)
 Main PID: 1613 (code=exited, status=1/FAILURE)

Jan 31 02:20:46 bananapi meyectl[1613]: File "/usr/lib/python2.7/subprocess.py", line 568, in check_output
Jan 31 02:20:46 bananapi meyectl[1613]: process = Popen(stdout=PIPE, *popenargs, **kwargs)
Jan 31 02:20:46 bananapi meyectl[1613]: File "/usr/lib/python2.7/subprocess.py", line 711, in __init__
Jan 31 02:20:46 bananapi meyectl[1613]: errread, errwrite)
Jan 31 02:20:46 bananapi meyectl[1613]: File "/usr/lib/python2.7/subprocess.py", line 1327, in _execute_child
Jan 31 02:20:46 bananapi meyectl[1613]: raise child_exception
Jan 31 02:20:46 bananapi meyectl[1613]: OSError: [Errno 2] No such file or directory
Jan 31 02:20:46 bananapi systemd[1]: motioneye.service: main process exited, code=exited, status=1/FAILURE
Jan 31 02:20:46 bananapi systemd[1]: Unit motioneye.service entered failed state.
Jan 31 02:20:46 bananapi systemd[1]: motioneye.service failed.

Output from journalctl:

Jan 31 02:20:44 bananapi systemd[1]: Starting motionEye Server...
-- Subject: Unit motioneye.service has begun start-up
-- Defined-By: systemd
-- Support: http://lists.freedesktop.org/mailman/listinfo/systemd-devel
--
-- Unit motioneye.service has begun starting up.
Jan 31 02:20:44 bananapi polkitd[576]: Unregistered Authentication Agent for unix-process:1607:1254671 (system bus name :1.38, object path
Jan 31 02:20:46 bananapi meyectl[1613]: INFO: hello! this is motionEye server 0.38
Jan 31 02:20:46 bananapi meyectl[1613]: Traceback (most recent call last):
Jan 31 02:20:46 bananapi meyectl[1613]: File "/usr/bin/meyectl", line 9, in <module>
Jan 31 02:20:46 bananapi meyectl[1613]: load_entry_point('motioneye==0.38', 'console_scripts', 'meyectl')()
Jan 31 02:20:46 bananapi meyectl[1613]: File "/usr/lib/python2.7/site-packages/motioneye/meyectl.py", line 255, in main
Jan 31 02:20:46 bananapi meyectl[1613]: server.main(arg_parser, sys.argv[2:], command[:-6])
Jan 31 02:20:46 bananapi meyectl[1613]: File "/usr/lib/python2.7/site-packages/motioneye/server.py", line 435, in main
Jan 31 02:20:46 bananapi meyectl[1613]: run()
Jan 31 02:20:46 bananapi meyectl[1613]: File "/usr/lib/python2.7/site-packages/motioneye/server.py", line 359, in run
Jan 31 02:20:46 bananapi meyectl[1613]: test_requirements()
Jan 31 02:20:46 bananapi meyectl[1613]: File "/usr/lib/python2.7/site-packages/motioneye/server.py", line 260, in test_requirements
Jan 31 02:20:46 bananapi meyectl[1613]: has_motion = motionctl.find_motion()[0] is not None
Jan 31 02:20:46 bananapi meyectl[1613]: File "/usr/lib/python2.7/site-packages/motioneye/motionctl.py", line 58, in find_motion
Jan 31 02:20:46 bananapi meyectl[1613]: binary = subprocess.check_output(['which', 'motion'], stderr=utils.DEV_NULL).strip()
Jan 31 02:20:46 bananapi meyectl[1613]: File "/usr/lib/python2.7/subprocess.py", line 568, in check_output
Jan 31 02:20:46 bananapi meyectl[1613]: process = Popen(stdout=PIPE, *popenargs, **kwargs)
Jan 31 02:20:46 bananapi meyectl[1613]: File "/usr/lib/python2.7/subprocess.py", line 711, in __init__
Jan 31 02:20:46 bananapi meyectl[1613]: errread, errwrite)
Jan 31 02:20:46 bananapi meyectl[1613]: File "/usr/lib/python2.7/subprocess.py", line 1327, in _execute_child
Jan 31 02:20:46 bananapi meyectl[1613]: raise child_exception
Jan 31 02:20:46 bananapi meyectl[1613]: OSError: [Errno 2] No such file or directory
Jan 31 02:20:46 bananapi systemd[1]: motioneye.service: main process exited, code=exited, status=1/FAILURE
Jan 31 02:20:46 bananapi systemd[1]: Unit motioneye.service entered failed state.
Jan 31 02:20:46 bananapi systemd[1]: motioneye.service failed.

ccrisan, 2018-01-31:
Do you not have the `which` command in your $PATH?

jwc866, 2018-02-01:
Apparently the "which" program was not installed on this CentOS image for ARM.  After installing it, the service starts successfully now.  Thank you for all of your help.

ccrisan, 2018-02-01:
Make sure you have mime types installed on your system.Particularly, python looks for `/etc/mime.types`.

closed 2018-02-01 by ccrisan