motioneye-project/motioneye #782: Unable to FTP images from Motioneye to FTP server

gbakakos1, 2018-01-31:
At this point I ran a test and I'm unable to get my images or video saved to my FTP directory.  Settings are below.

[image]

Folder get's created but nothing seems to be directed to it.   
[motion (1).log](https://github.com/ccrisan/motioneye/files/1683140/motion.1.log)
[motioneye (3).log](https://github.com/ccrisan/motioneye/files/1683142/motioneye.3.log)
[messages (1).log](https://github.com/ccrisan/motioneye/files/1683145/messages.1.log)
 Any help would be appreciated.

ccrisan, 2018-02-01:
There are many FTP errors in your log. Some of them may be due to using backslashes in Location. I strongly recommend using only forward slashes. Backslashes are usually used as escape characters and god knows what they do when used in an FTP command.

eserk, 2019-07-27:
I used these settings and I received a successful confirmation, but when I try to use ftp, it will deny me access.
[image]

eserk, 2019-07-27:
[image]

jasaw, 2019-07-29:
You are using smb protocol rather than ftp in your "Location" config item.

eserk, 2019-07-29:
Yeah, it succeeds but does not transfer the files to my open media vault shared folder, I am going to put my credentials in and send the logs here, I cannot figure out what I am doing wrong. 
[motioneye(1).log](https://github.com/ccrisan/motioneye/files/3440098/motioneye.1.log)

eserk, 2019-07-29:
[image]
[image]
[image]

eserk, 2019-07-29:
[OMV syslog.zip](https://github.com/ccrisan/motioneye/files/3440121/OMV.syslog.zip)

eserk, 2019-07-29:
Nevermind, I figured it out, as soon as the include Subfolders was turned off, everything worked, I guess I was getting some error due to permissions, even though just about everything was set to explicit allow, weird.

bobster316, 2020-02-21:
In which section were the subfolders turned off?

Yonny24, 2020-11-25:
I'm still struggling with this also. Using the addon.
ftp share on an asus router. Ipsy agent dvr will upload fine but no matter what settings I try (hours attempting).
I get no success. various errors. permission issues/name resolution issues.

Almost ready to quite motioneye and just use agent dvr

starbasessd, 2020-11-25:
In order to help, we need specifics:
motioneye.log, settings for connection, etc.
Are you using motion/motioneye on another OS (which OS?) or motionEyeOS? (which install file used?) Which install instructions did you use?
If you are using motionEyeOS, it would be better to open a new issue at 
https://github.com/ccrisan/motioneyeos/issues
and answer as many questions as possible, and attach all the logs requested...

Yonny24, 2020-11-25:
Hi.

I'm using the motioneye addon.
https://github.com/hassio-addons/addon-motioneye
homeassistant supervised on ubuntu

Been searching and asking for weeks now. No reply from the owner of the addon yet.
Very puzzling indeed. Must be something simple but have missed or even a bug with the addon.
Agent dvr running as a docker container on my linux connects perfectly. Also the ipsy service on a windows 10.

Just this addon is being rejected or refusing to connect to the sae ftp share.

starbasessd, 2020-11-25:
The owner of the add-on has never answered me as to the changes they made,
and there are many things broken.
Have you tried the docker version of motionEye?
https://github.com/ccrisan/motioneye/wiki/Install-In-Docker
Have you tried the Ubuntu version of motionEye?
https://github.com/ccrisan/motioneye/wiki/%28Install-On-Ubuntu-%2820.04-or-Newer%29
Let me know if you'd like support in those.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Yonny24, 2020-11-25:
> The owner of the add-on has never answered me as to the changes they made, and there are many things broken. Have you tried the docker version of motionEye? https://github.com/ccrisan/motioneye/wiki/Install-In-Docker Have you tried the Ubuntu version of motionEye? https://github.com/ccrisan/motioneye/wiki/%28Install-On-Ubuntu-%2820.04-or-Newer%29 Let me know if you'd like support in those.
> […](#)
> -- Thanks Kevin Shumaker Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel> N38° 19' 56.52" W85° 45' 8.56" Semper Gumby “Don't tell people how to do things. Tell them what to do and let them surprise you with their results.” - G.S. Patton, Gen. USA Ethics are what we do when no one else is looking. Quis custodiet ipsos custodes? “There is no end to the good you can do if you don’t care who gets the credit.” - C Powell You know we're sitting on four million pounds of fuel, one nuclear weapon and a thing that has 270,000 moving parts built by the lowest bidder. Makes you feel good, doesn't it?

Great might just try one of these instead of the addon. Which do you recommend? Possibly the docker container? Easy to manage through portainer then.

starbasessd, 2020-11-25:
Don't know how it/they would interact with your other software.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Yonny24, 2020-11-25:
Just ran up a docker image of motioneye. Added a camera and tried the ftp upload. Identical log err3. Still could not progress. Stopped the container and reverted back to the addon for now.
May need to install the ispy on Ubuntu as I have it on windows and works as expected.
Thanks

starbasessd, 2020-11-25:
Try adding -p 21:21 to your startup for the docker for grins and giggles...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Yonny24, 2020-11-26:
Considering getting a raspberry pi and using it solely for motioneyeos to free resources from my ha machine.

starbasessd, 2020-11-26:
Docker has it's own firewall, and may not be allowing ftp traffic through,
hence adding the -p 21:21 to the startup. Just like there is the -p
8765:8765 to allow the WebGUI traffic in and out...

Yonny24, 2020-11-26:
Great I'll try this.
Is this port rule just for this container or at the docker level? Sorry maybe a noob question. 
How do I go about this with the portainer container terminal?

starbasessd, 2020-11-26:
OK, installed docker on a Pi4-4GB sitting idle
Used the following command line to install and start motioneye:
sudo docker run --name="motioneye"     -p 8765:8765     -p 21:21
--hostname="motioneye"     -v /etc/localtime:/etc/localtime:ro     -v
/etc/motioneye:/etc/motioneye     -v /var/lib/motioneye:/var/lib/motioneye
 -v /var/log:/var/log   --restart="always"     --detach=true
ccrisan/motioneye:master-armhf
(all on one line)
Set up "Upload Media" to my  NAS FTP server:
Server: [ip_address]
Server port: 21
Location: [/ftp_share/drive_share/folder]
Include subfolders ON
username and password
Clicked 'Apply' at top
Click test
Works.
Changed Image file name to just time [/%H-%M-%S for testing]
clicked the camera icon in the image from camera, waited a few seconds (<4)
appeared.

On Wed, Nov 25, 2020 at 8:18 PM Kevin Shumaker <kevin.shumaker@gmail.com>
wrote:

> Docker has it's own firewall, and may not be allowing ftp traffic through,
> hence adding the -p 21:21 to the startup. Just like there is the -p
> 8765:8765 to allow the WebGUI traffic in and out...
>
>
>

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-11-26:
No clue. I don't use dockers normally, never needed to before supporting
motionEye...The command line I just sent is just for the motioneye
container, not generically, AFAIK...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Yonny24, 2020-11-26:
> 
> OK, installed docker on a Pi4-4GB sitting idle Used the following command line to install and start motioneye: sudo docker run --name="motioneye" -p 8765:8765 -p 21:21 --hostname="motioneye" -v /etc/localtime:/etc/localtime:ro -v /etc/motioneye:/etc/motioneye -v /var/lib/motioneye:/var/lib/motioneye -v /var/log:/var/log --restart="always" --detach=true ccrisan/motioneye:master-armhf (all on one line) Set up "Upload Media" to my NAS FTP server: Server: [ip_address] Server port: 21 Location: [/ftp_share/drive_share/folder] Include subfolders ON username and password Clicked 'Apply' at top Click test Works. Changed Image file name to just time [/%H-%M-%S for testing] clicked the camera icon in the image from camera, waited a few seconds (<4) appeared.
> […](#)
> On Wed, Nov 25, 2020 at 8:18 PM Kevin Shumaker ***@***.***> wrote: Docker has it's own firewall, and may not be allowing ftp traffic through, hence adding the -p 21:21 to the startup. Just like there is the -p 8765:8765 to allow the WebGUI traffic in and out...
> -- Thanks Kevin Shumaker Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel> N38° 19' 56.52" W85° 45' 8.56" Semper Gumby “Don't tell people how to do things. Tell them what to do and let them surprise you with their results.” - G.S. Patton, Gen. USA Ethics are what we do when no one else is looking. Quis custodiet ipsos custodes? “There is no end to the good you can do if you don’t care who gets the credit.” - C Powell You know we're sitting on four million pounds of fuel, one nuclear weapon and a thing that has 270,000 moving parts built by the lowest bidder. Makes you feel good, doesn't it?

hmm odd - followed your instructions to the T. container created.
after clicking test:
'Accessing the upload service failed: 550 Permission denied (restricted operation).!'

Thanks

starbasessd, 2020-11-26:
What kind of FTP server did you connect to?
I used
sudo apt install docker.io -y [enter]
for my docker on my Pi4-4GB. (moved USB stick to a Pi4-8GB, same stick
works fully there, too)

https://www.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=&cad=rja&uact=8&ved=2ahUKEwjRw6ivx6DtAhUNxVkKHTTZDSEQFjAAegQIDBAC&url=https%3A%2F%2Fserverfault.com%2Fquestions%2F598392%2Fftp-550-permission-denied&usg=AOvVaw3fDx5dcBvMqjWdUYm3z_pm

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Yonny24, 2020-11-26:
> 
> What kind of FTP server did you connect to? I used sudo apt install docker.io -y [enter] for my docker on my Pi4-4GB. (moved USB stick to a Pi4-8GB, same stick works fully there, too) 

Just a simple micro sd card in a usb adaptor at the back of the router. set up on the ftp share function on the asus router.
Everything else uploads to it fine. (ipsy agent dvr etc.)

starbasessd, 2020-11-26:
I don't know.
It's working for me with my described setup in the plain docker image and
the startup I posted. I'm using a Buffalo NAS with FTP & SMB v1, and it's
communicating both with 3 D-Link and 1 Foscam camera directly, an Ubuntu
20.10 / motion / motionEye Lenovo hub, a Pi4-4GB motionEye Docker, and
sever VMs in VirtualBox.
I've also built a VM with Ubuntu and vsftp and it works there.
Maybe someone else has ideas...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Yonny24, 2020-11-26:
Thanks for helping.Would be good to get to the bottom of the ftp access issue.Although this has pushed me to buy a NVR. If the HA ever goes down while away I have the dedicated NVR to continue monitoring as we are in a country with a high risk of break ins.

 
 

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-11-26:
Be sure to get a UPS for power backup.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Yonny24, 2020-11-26:
> 
> Be sure to get a UPS for power backup.
> […](#)

Yep could do with one for the main gateway router and this for sure. Not cheap though
Alarm system has one and sim to send signals to the security company.

starbasessd, 2020-11-26:
Car battery, trickle charger, and small inverter... Poor Man's UPS (in use
in my storage shed out back) Harbor freight solar panels, the controller
they come with, a motorcycle battery and a small inverter in my
cargo/camping trailer. Both under $200 total (batteries were used/recycled)

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Yonny24, 2020-11-26:
Unfortunately no cheap items are cheap here. no ebay/walmart/amazon. Very cheap chinese items are expensive.
the inverters are not cheap neither are car batteries. Got an old car battery but its pretty dead showing 2volts.

JamBalaya56562, 2026-11-08:
Closing, as this was with the Docker image of 2018, and the later replies went off topic. The first problem was the backslashes in Location, which should be forward slashes as ccrisan said. The second was an `smb://` address in an FTP upload. Please open a new issue with `motioneye.log` if FTP uploads still fail with a current image.

closed 2026-11-08 by JamBalaya56562