motioneye-project/motioneye #783: I was only able to add a USB camera as first camera, before RPI cam

phisher1, 2018-01-31:
When attempting to add a secondary USB web camera, which is clearly detected as a v4l2 device, motioneye would not detect any cameras.

When I removed the RPI camera and then started over with no cameras, only then was I able to add the USB camera.

Jan 31 17:08:27 printerpi meyectl[693]:    DEBUG: listing V4L2 devices
Jan 31 17:08:27 printerpi meyectl[693]:    DEBUG: found device Microsoft® LifeCam Studio: /dev/video0, /dev/v4l/by-id/usb-Microsoft_Microsoft®_LifeCam_Studio_TM_-video-index0

ccrisan, 2018-02-01:
1. When you have both cameras connected, how do you know motionEye does not detect the second one? What camera type(s) have you selected in the Add Camera dialog?
2. When you have both cameras connected, what does the following command say:

        v4l2-ctl --list-devices

phisher1, 2018-02-03:
Through the motioneye UI, if the RPI camera was already added, then went to Add Camera,  select "Local V4L2 Camera", it would tell me (no cameras). 

When I removed the RPI camera, so there were no cameras, then Add Camera, select "Local V4L2 Camera", my USB camera would show up.

With both cameras connected:
pi@printerpi:~ $  v4l2-ctl --list-devices
Microsoft® LifeCam Studio(TM) (usb-3f980000.usb-1.4):
	/dev/video0

ccrisan, 2018-02-05:
@phisher1 make sure your system detects two cameras. More precisely, `v4l2-ctl --list-devices` should detect two cameras. Otherwise motionEye cannot do anything about it.

phisher1, 2018-02-05:
I'm confused.  My RPI camera is never detected as a V4L2 Camera, it is MMAL.

ccrisan, 2018-02-06:
@phisher1 MMAL is just as fine (or maybe even better). So what you're saying is that after you add your MMAL camera to motionEye, you can't add your second (V4L2) camera because it's not listed in the Add dialog?

If that's the case, is this reproducible? Can you reproduce it with a clean configuration? Also what motionEye version do you run?

phisher1, 2018-02-18:
Sorry for not getting back to this sooner. I made a video demonstrating.  Don't mind the camera not showing up.. (gray screen) it normally does =)  .. that's another thing I'm troubleshooting. It just loses connection someitmes.  Anyway..
https://www.youtube.com/watch?v=wDxyP9GmvrE

ccrisan, 2018-02-21:
@phisher1 I get it now. It's indeed a problem that might have been fixed in 0.38.1. Can you please try it and see if the problem still persists?

phisher1, 2018-02-21:
Can you update to 0.38.1 in PyPI ?

ccrisan, 2018-02-23:
Done!

phisher1, 2018-02-23:
And confirmed working. Awesome, Thank you! 
Motioneye is fantastic. I use it on 3 different raspberry pi's currently. I've used it many more in the past. My brother uses it as well. Well done.

closed 2018-02-23 by phisher1

perlian, 2021-12-09:
I have exact the same problem with actual version 042.1 !!

starbasessd, 2021-12-09:
@perlian Please open a new issue with the following info:
Board being used
OS being used
Which set of instructions to set up motionEye being used
Which USB cameras being used
We will be glad to try to help.