motioneye-project/motioneye #784: [Errno 13] Permission denied: '/home/pi/myNAS/Surveillance/Voordeur/.keep'

HWL1968, 2018-02-03:
Hi all,

I installed motioneye and motion from scratch on a RPi2. System runs fine, uses 2 IP cameras and recordings on the SD card can be done without any issue. I want the files to be stored on a NAS (tried FTP: didn't manage to get it working: could trace on the NAS that the session was opened, but nothing was written. anyhow).

When I switch to samba sharing: I change the configuration into 'custom path': /home/pi/myNAS/Surveillance/Voordeur/. 

Recording is working fine, playback also. However: when I try to delete the files that have been recorded, it only succeeds once. A .keep file has then been created in the /../Voordeur/ directory whereafter none of the files can be deleted anymore [Errno 13] Permission denied: '/home/pi/myNAS/Surveillance/Voordeur/.keep'. 

Removing the .keep file by hand is no problem and then the removal of the files works again for 1 file or 1 subdirectory (whatever comes first). 

I have read about the introduction of the .keep file, which is also being created on the SD card but there it is working as it should (no issues).

Help??

ccrisan, 2018-02-05:
> I want the files to be stored on a NAS (tried FTP: didn't manage to get it working: could trace on the NAS that the session was opened, but nothing was written. anyhow).

Well we need to somehow get this FTP working. It's working for virtually everybody else so there must be a little something that needs to be adjusted or fixed.

> A .keep file has then been created in the /../Voordeur/ directory whereafter none of the files can be deleted anymore [Errno 13] Permission denied: '/home/pi/myNAS/Surveillance/Voordeur/.keep'.

The `.keep` file is there to prevent removing the *root directory* of your camera (`target_dir` in motion config terms). It should be there, assuming that `/home/pi/myNAS/Surveillance/Voordeur/` is indeed your root directory. However you shouldn't get a permission denied error, unless you have some permission issues across the NAS setup. Can you try removing it manually *using the user under which motionEye runs* (which is what, btw)?

ccrisan, 2018-02-05:
Also the path you mentioned earlier, `/../Voordeur/`, doesn't make any sense in this context. Where have you gotten that from? Maybe sharing your configuration with us would shed some light on the matter here.

HWL1968, 2018-02-05:
Hi Calin,

Regarding your questions:

- FTP: I agree: it is probably something silly with the NAS. I did try to trace on the NAS what is happening, but the NAS only shows that a connection is being setup, directory is being changed, then connection is broken, no attempt to write. FTP'ing by hand, shows that the FTP user is able to write and delete.. Very confusing. I know, I am crying for myself: feeling lonely somehow. :-)

- .keep: as mentioned, I found your post which describes the need for the .keep file. I think it is working excellent in general, since there isn't an issue when writing to the SD card (this same configuration with 1 change: the root directory). Changing however from SD to SMB (mounted in a directory as written) results in this behaviour. 

Regarding the directory:
I log in to the system as pi, hence I use /home/pi as my home directory. In this directory I created a directory myNAS in which is the mountpoint of the SMB directory /Surveillance/
Path sofar: /home/pi/myNAS/Surveillance/

I have added an IP camera to watch the frontdoor ('Voordeur'). I store all the videofiles into this directory: /home/pi/myNAS/Surveillance/Voordeur/... 
The directory structure is then divided into: yyyymmdd/hh/ in which the actual recordings are being placed.

The user motion(eye) is running under is (as far as I guess) 'motion' (did an apt-get install). Trying 'sudo -u motion rm .keep' removes the file. (file permissions are 777 - rwx for all -).

What configuration do you need? I try to sum it up:
- RPI2
- Foscam FI9900P cam (IP cam)
- NAS: Asustor AS6204T
- mounted the share at  /home/pi/myNAS/
- did an apt-get install of motion / motioneye
- base is the newest Raspbian 2017-11-29
- added to fstab: 
//192.168.1.xx/Surveillance /home/pi/myNAS/Surveillance cifs username=<user>,password=<password>,rw,file_mode=0777,dir_mode=0777 0 2

I double checked via FTP if the user is able to delete / write files: yep, is working (as one can also see if the .keep file isn't there yet)

The path mentioned: I inserted that in the motioneye menu: menu 'file storage', item: 'Root directory'. 

What files do you need, I will upload them tomorrow. I am sure I did some config wrong, but can't find it.

ccrisan, 2018-02-06:
> I store all the videofiles into this directory: /home/pi/myNAS/Surveillance/Voordeur/...

What's with the trailing dots? I'm assuming two dots for *parent directory* and one for ending the sentence. Or? In any case, those probably shouldn't be there. 

> The user motion(eye) is running under is (as far as I guess) 'motion'.

Are you 100% sure of that? By default both motionEye and motion run as root, unless you have changed the way you start motionEye.

Otherwise your setup looks fine to me.

HWL1968, 2018-02-07:
Sorry for the confusion I created: I meant with the dots that subdirectories will follow (and are generated by motion while recording). They aren't in the configuration. Also no dots for the parent, the path is declared from the root within the RPi.

Regarding the user: I assumed 'motion' but don't know, since I installed using apt-get and didn't change anything to be sure things will work as designed. How can I check?

HWL1968, 2018-02-07:
Just to add: removal of the videofiles work fine as mentioned. When I remove the .keep file by hand, it works again (be it, just once, then the .keep is created again).

ccrisan, 2018-02-08:
At this point, I'm out of ideas. I'm willing to log in remotely to your motionEye system and try to identify the problem myself, if that's ok with you. If yes, please contact me via email.

HWL1968, 2018-02-14:
Sorry for the late respons. Thx for your offer, but I will try to solve it by hand for now. using a cron to remove the older files. I still do not understand what the issue is, I assume it is probably the NAS but anyhow. Thx for your offer, I will close the issue. If I succeed in solving, I will post the 'lesson learned'.

closed 2018-02-14 by HWL1968

marrobHD, 2019-06-29:
Sameissue. @ccrisan you can connect to my pc if you want

GevaudanBeast, 2020-10-12:
Sameissue. @ccrisan it's possible to modify possibilities script to remove .keep ?