motioneye-project/motioneye #790: Google Chrome - flickering scrollbar

dbruneau-ie, 2018-02-08:
Hi,

I've been experiencing a small cosmetic issue when monitoring cameras with Google Chrome.

I'm having a hard time describing with words so here is a gif..

[image]

What I'm showing is

- No overlay, no scrollbar
- Enable overlay, scrollbar appears (which is fine) but then proceeds to flicker.  
- Disable CSS rule(?) for "margin" on the item and flickering ceases.

From what I can tell, it flickers anytime the framerate information is updated.   I played around with the CSS for the camera-info and it turns out that disabling "margin: 0px" removes the scrollbar flickering.

Windows
Chrome (any version I've had in the last 2-3 years).
I can't replicate in Firefox.

Thanks in advance,

ccrisan, 2018-02-08:
I must admit this sucks. I'll look into it and see if I can reproduce it myself.

dbruneau-ie, 2018-02-09:
Thanks!

I skimmed through the Chromium bug tracker to see if I could find a ticket that could explain what's happening but came up empty handed.

thehaxxa, 2020-06-26:
I also experience this in Google Chrome & Brave Browser, particularity when the resolution of the cameras match the resolution of the Browser Monitor.

JamBalaya56562, 2026-11-15:
I tried this with the current version in Chrome 156: with the frame overlay shown, changing the frame rate text no longer changes the size of the page, the camera frame or the overlay, since the overlay now sits on top of the frame. At window sizes where the page is just as high as the window, the scrollbar did not appear or disappear while the frame rate was updating. Closing this, as it no longer happens; if you still see the flickering with motionEye 0.44 or later, please reopen with the browser version and the window size.

closed 2026-11-15 by JamBalaya56562