motioneye-project/motioneye #793: Need to edit motioneye.conf but I forget how to ssh in

Billium28, 2018-02-09:
I want to change the smb share to true but forget how to use SSH to change the value. I had done it before but haven't needed to do much with it in a long time.

I tried 

nano /etc/motioneye/motioneye.conf
sudo nano /etc/motioneye/motioneye.conf
it opens an empty file and says directory /motioneye does not exist.

ccrisan, 2018-02-09:
What board and OS? How did you install motionEye?

Billium28, 2018-02-09:
Ah, sorry left out the important stuff.

Raspberrypi Zero W, I have the network setup.
I realize now it was the MotioneyeOS install.
I installed with the Raspberrypi 1 (original) file on the MotioneyeOS site. -
https://github.com/ccrisan/motioneyeos/releases

I wrote it to the sd card with Etcher. I am now wondering if I made a mistake in installing the OS, though I can log into the motioneye on the pi fine and it is working.

I use motioneye docker on my Unraid server to access the 2 pi's I have running. I had them working for months but in Dec. I noticed they had stopped uploading (because of an Unraid docker issue.) I reinstalled the motioneye OS on both today to just start fresh.

kleini, 2018-02-10:
If you want to have latest motioneye release inside docker, too, choose the ccrisan/motioneye:stages-amd66/amrhf image. Unfortunately docker on Travis CI was upgraded and latest build from dev and master branches did not upload.

ccrisan, 2018-02-12:
On motionEyeOS you have /data/etc/motioneye.conf

closed 2018-02-12 by ccrisan