motioneye-project/motioneye #794: Very Good Software, just one issue with pi cam locally

bend94, 2018-02-10:
Hi,

I can add ip cam working perfect.
I plug a pi cam locally on the motioneye server locally

root@raspberrypi.3:motion # vcgencmd get_camera
supported=1 detected=1

Now in motioneye, i added it 

local V4L2 Camera and motion eye propose me : "mmal service 16:1"
i clicked add
But the camera windows stay grey with the message unable to open  video device all is grey

the device /dev/video0 exist !
Please help
thanks

bend94, 2018-02-11:
hi again, my picam 2.1 (bought in amazon) is not working neither with Raspbian nor eyemotion os.
and neither with pi3 nor pi zero.
it is strange because the cam is seing by the command vcgencmd in all my tests (raspbian, motioneyeos, pi3, pi zero)

root@raspberrypi.3:motion # vcgencmd get_camera
supported=1 detected=1

but gray windows in motioneye and "Unable to open video device" even if /dev/video0 is there

please help
regards

jasaw, 2018-02-12:
Try raspistill and raspivid on Raspbian. If camera is not working, it may be broken. There's a tiny rectangular connector just next to the camera lens. Make sure that hasn't come off.

bend94, 2018-02-12:
hi

Raspistill is giving an error : 

raspistill -v -o test.jpg

raspistill Camera App v1.3.8

Width 2592, Height 1944, quality 85, filename test.jpg
Time delay 5000, Raw no
Thumbnail enabled Yes, width 64, height 48, quality 35
Link to latest frame enabled  no
Full resolution preview No
Capture method : Single capture

Preview Yes, Full screen Yes
Preview window 0,0,1024,768
Opacity 255
Sharpness 0, Contrast 0, Brightness 50
Saturation 0, ISO 0, Video Stabilisation No, Exposure compensation 0
Exposure Mode 'auto', AWB Mode 'auto', Image Effect 'none'
Metering Mode 'average', Colour Effect Enabled No with U = 128, V = 128
Rotation 0, hflip No, vflip No
ROI x 0.000000, y 0.000000, w 1.000000 h 1.000000
mmal: mmal_vc_component_enable: failed to enable component: ENOSPC
mmal: camera component couldn't be enabled
mmal: main: Failed to create camera component
mmal: Failed to run camera app. Please check for firmware updates

Regards

bend94, 2018-02-12:
hi-again
i saw with my provider (amazon) 
i can resend back the out of order cam and be reemboursed

i plan to buy a new one, what cam is the best choice to be able to manager it with raspberry zero and motioneye os ?

regards

chrisf4lc0n, 2018-02-17:
Have you enabled the camera through the raspi-config and have you given your system at least 64MB VRAM?

JamBalaya56562, 2026-11-07:
Closing, as `raspistill` failed outside motionEye as well, which points to the camera itself, and you were returning it for a new one.

closed 2026-11-07 by JamBalaya56562