motioneye-project/motioneye #795: Higher CPU utilization with Raspbian than MotionEyeOS

rodelco, 2018-02-10:
Hello everyone;  More of a question than an issue:

I am pretty new to Linux, GitHub, Raspberry Pi, etc., but am having fun trying to learn.  I first wanted to thank all of those involved with this MotionEye project.  Obviously a lot of work and talent has gone into it.  I am a retired engineer who is trying to set up some home surveillance cameras, which led me here.  

As I said, I am a complete NOOB, so forgive me if this has been asked before, or if it is a stupid question.  I currently am using a RPI 3B as my MotionEye "server", and I have a Wansview W2 IP Camera and a Raspberry Pi Zero W with camera connected to this server via Wi-Fi.  The "server" is hardwired to my network.  

Using the "top" command on the server, I notice that when I use MotionEyeOS, the CPU utilization is running around 70% or so, but when I run MotionEye on a Raspbian Stretch OS, the same setup shows a CPU utilization around 130% or so.

Can someone explain why the "motion" program would use more CPU time on Raspbian vs MotionEyeOS?  

Thank you for any help and guidance you can offer!

jasaw, 2018-02-12:
Different versions of motion?
One uses hardware accelerated video encoding, the other doesn't?

rodelco, 2018-02-12:
**jasaw:** Well that is certainly a good thought.

  Let me check....It looks like Motion is Version 4.1.1 on my Raspbian 9.3 install, and Motion is Version 4.0.1+git2a140fa (I don't know what the "+git2a140fa" means), on the MotionEyeOS install (OS Version is 20171008).

Thanks for your reply.

jasaw, 2018-02-12:
`git2a140fa` is the git hash of that version. https://github.com/Motion-Project/motion/commit/2a140fa31feafd960033ad292615a601a1f3faec

chrisf4lc0n, 2018-02-16:
What resolution/FPS have you set for the camera up?

rodelco, 2018-02-16:
I am experimenting with two cameras right now, and have been using the Raspbian Stretch installation.  I have a RPi standard camera connected to a Pi Zero W running MotionEyeOS and configured as a "Fast Cam".  I have the Video Resolution on the Camera's MotionEye set to 1024 x 576, with a Frame Rate of 20 and Image Quality set to 25.  On the MotionEye Server side, I have the Video Device FrameRate also set to 20.  RIght now this camera is displaying ~ 28/12 fps in the lower left corner.

I also have a Wansview W2 Wi-Fi IP camera that I am connecting to the MotionEye Server with its RTSP stream.  I am using the second stream from this camera (rtsp://192.168.xxx.xxx/live/ch1) which is set to a resolution of 640 x 360 and a frame rate of 25 fps.  ON the MotionEye Server side, I have the Video Device Frame Rate set to 15 and the Video Resolution to 640 x 360.  RIght now this camera is displaying ~16/11 fps in the lower left corner.

I am pretty happy with the video that I am getting, I was mainly concerned that the Pi3 that is running the MotionEye Server might be overworked.  Right now, the "top" command shows the %CPU running between 107% and 117% with both of these cameras streaming data.  I am only using motion detection on the Wansview camera right now. 

When I set these two cameras up the same way on the MotionEyeOS distro instead of the Raspbian Stretch distro, the CPU usage was much lower, so I thought maybe there was something I could do to improve (reduce) the CPU usage.  I have only been working with Raspbian since I started trying to learn about Linux and the RPi, so I am just more familiar with it then the "BuildRoot" distro that MotionEyeOS is built on.

Thanks.

chrisf4lc0n, 2018-02-16:
It might be a daft question, but is it the MotionEye/Motion process taking up most of the resources on the Pi3?

rodelco, 2018-02-16:
[image]

It looks like it is to me, but I am certainly not an expert in anything yet!

chrisf4lc0n, 2018-02-16:
CPU usage is actually 33.8% (third line from the top). ;)

rodelco, 2018-02-16:
OH, so why are the units after that number in microseconds??

What is the %CPU column telling me?  Sorry for the dumb questions, but thanks for your help!

chrisf4lc0n, 2018-02-16:
[https://www.maketecheasier.com/linux-top-explained/](https://www.maketecheasier.com/linux-top-explained/)

rodelco, 2018-02-16:
Thanks for the link.  So, if I understand it correctly, in my case 33.8% of the time, the CPU has been used on "USER" processes, and has actually been idle for 58.4% of the time.  The 107.6 %CPU number tells me that the "motion" process has used 107.6% of the CPU time since the last update, and since this is a quad-core device, that would mean that is is using a little more than 25% of the total CPU cores??

So, while looking into this, I just discovered the "IRIX" mode in the TOP command.  Is it possible that in the BuildRoot distro, that the IRIX mode is OFF by default and that is why the numbers looked so different??  I will have to re-load my MotionEyeOS image and check that out when I get a chance.  If that is true, then I will feel a lot better about my Raspbian installation of MotionEye!

Thanks so much for your help!

chrisf4lc0n, 2018-02-16:
Try htop, is more intuitive and easier to managed, similar to nano as opposing to vi.

chrisf4lc0n, 2018-02-18:
@ccrisan I think we can close this one down.

rodelco, 2018-02-18:
Yes, thank you for your help.  It appears to me that the differences I saw between the CPU loads on the different distros was due to my not understanding what the "top" program was telling me, as well as apparently there are different versions of the "top" program in the two distros.  

Thanks again.

closed 2018-02-18 by rodelco