motioneye-project/motioneye #797: Feature request: advanced mask file

richri, 2018-02-18:
Hi,

I would like to suggest an enhancement for the mask file dialog: Motion supports mask files as *.pgm files. Here, in general it is possible to add different grey levels to influence the motion detection: white means no change to the original image, whereas black means to ignore the specific pixel of the original image. The grey tones in between allow to reduce the influence on the motion detection.

I have certain parts in my camera setup that I do not want to exclude completely, bute would like to reduce the sensibility. Can such a feature be added in the masl file editor? Maybe some kind of 5-way toggle? 1st click --> black, 2nd click --> very dark grey, 3rd click --> grey, 4th click --> lighter grey, 5th click --> no grey value.

Alternatively an option to specify a manual uploaded mask file would be helpful.

Keep up the great work!

Best regards

Christian

ccrisan, 2018-02-19:
Is this feature supported by motion?

richri, 2018-02-19:
Yes, it is supported. The hint is hidden in the detailed description for the option, for example here: [http://www.lavrsen.dk/foswiki/bin/view/Motion/ConfigOptionMaskFile](http://www.lavrsen.dk/foswiki/bin/view/Motion/ConfigOptionMaskFile)

> The areas that you want detected must be white. The error that you want ignored must be black. The pgm image must be the same size (number of pixels high and wide) as the pictures that are taken by the camera (video4linux device).
> 
> You can adjust sensitivity by using gray tones. 

It really is a great feature to fine tune sensitivity of detection in certain areas of the picture.

ccrisan, 2018-02-22:
I did not know about that. However having such a complex mask editor in motionEye seems quite difficult to implement. I personally am not going to implement it and I don't think it would bring too many benefits compared to what we currently have.

closed 2018-02-22 by ccrisan

richri, 2018-02-22:
How about an alternative then? Providing a possibility to upload a custom *.pgm file instead of using the one generated by your editor?

By now, it is only possible to set a file via your editor, replace it via commandline on the server and never touch the dialog again. As soo as the edit mask file button is clicked again, the custom file is overwritten.

TopperBG, 2018-02-23:
Sounds very reasonable.

ccrisan, 2018-02-24:
@richri you're right. There must be a way to prevent the UI from overwriting externally customized options. However, the current motionEye architecture does not allow us to easily implement it. I'll have to think about a general way to do that.

bertdebondt, 2019-03-10:
Maybe a dumb question, but it there a possibility to change the grid size (rectangles) of the mask ? I would like to use a smaller rectangle size in the MotionEye web GUI to select/deselect mask area. Thanks.