motioneye-project/motioneye #799: Edit despeckle filter

richri, 2018-02-19:
Hi,

when I want to enable and set another despeckle filter that the default option, I can add `despeckle_filer eeedddl` for example in the Extra motion options. When I apply, they take effect, but when I call the gui again later, I do not see any hint that the despeckle_filter is changed from the default. the extra motion options filed is empty now.

Could you add a way so that you can actually see the customized filter?

Best regards

ccrisan, 2018-02-22:
The despeckle filter is managed internally by motionEye. The default value is `EedDl`. You can change it in Extra Motion Options but it won't be displayed in the UI afterwards. Its value however shouldn't be lost.

I may add a specific despeckle filter UI option in the future but, given the complexity behind this field, I don't think many users would benefit of it.

closed 2018-02-22 by ccrisan

richri, 2018-02-22:
The only problem that I see is, that I can add the extra option via your GUI, but as soon as I save and reload the GUI I do not see this customized param any more. So i Have to know, that some weeks I set an extra option ...

Maybe you could keep track of extra options added there by saving them comma separated in a list, so that when reloading the GUI you could display these params in the extra-options field again.