motioneye-project/motioneye #800: Feature Request: Toggle motion detection on/off via API or else

rafik24, 2018-02-20:
Hi Ccrisan,

I use motion eye and i would like to enable motion detection and recording only when i arm my home alarm and would need a way to do this in motioneye.

I guess i could stop the service and shuffle files around but i want to make sure i am not too intrusive.

Thanks,
Rafik

formula0ne, 2018-03-15:
I second this, could not find any way to do this! Or is there a way?

jasaw, 2018-03-15:
A more elegant way is to implement a REST API in motion software to support this. Try convincing motion developers to implement this.

janpascal, 2018-06-04:
There is a way - motioneye runs motions, which has a web control port (you can find the port number in the motion.conf configuration file, usually located in /etc/motioneye/motion.conf). To start motion detection, run a command like

`wget -O- "http://192.168.1.14:7999/1/detection/start" >/dev/null` 

The '1' is for the camera number.

To stop (pause) motion detection, run something like

`wget -O- "http://192.168.1.14:7999/1/detection/pause" >/dev/null` 

You can use a web browser pointed at `http://192.168.1.14:7999` to interactively control the motion instance used by motioneye.

schneivo77, 2018-12-17:
> There is a way - motioneye runs motion, which has a web control port (you can find the port number in the motion.conf configuration file, usually located in /etc/motioneye/motion.conf). I use it from my home alarm system to only run motion detection when the alarm system is armed, like the thread starter suggests.
> 
> To start motion detection, run a command like
> 
> `wget -O- "http://192.168.1.14:7999/1/detection/start" >/dev/null`
> 
> The '1' is for the camera number.
> 
> To stop (pause) motion detection, run something like
> 
> `wget -O- "http://192.168.1.14:7999/1/detection/pause" >/dev/null`
> 
> You can use a web browser pointed at `http://192.168.1.14:7999` to interactively control the motion instance used by motioneye.

This doesn't work for me.
I tried it with CURL and Postman and it seems that there is no server running on that port. I used the IP from the Raspberry Pi on which MotionEyeOS is running and the port which I found in the file `/data/etc/motion.conf`.

Am I missing anything?

kabadisha, 2019-01-11:
I have created a script that runs on my router and can detect if my phone is connected to the wifi network. If it is, it disables motion detection using the web control method above.

You can find the code here: https://gist.github.com/kabadisha/a87e50311d3ed4589f17a6e37c574a44

tobiaslohr, 2019-01-21:
Hi @ccrisan, 
I'm sure the topic is not new to you. Are there any thoughts about adding an API to MotionEye to unlock integrations into (open) Home Automation systems?

schneivo77, 2019-01-22:
> @schneivo77,
> I had the same issue. It seems that by default the web control port is only accessible, literally from **localhost** which is literally the machine motioneyeos is running on (a raspberry pi in my case).
> 
> To enable access to that port from other machines, you have to ssh onto the device and edit _/data/etc/motion.conf_
> Change `webcontrol_localhost on` to `webcontrol_localhost off` and then reboot the camera.
> 
> After that, you should be able to access the URLs :-)
> 
> It's a bit confusing in my opinion. I think it would be more clear if the property was called something like `webcontrol_restrict_to_localhost`.

Now it works! Nice! Thanks a lot! :)

hamiltont, 2019-06-09:
Note from Motion documentation (https://motion-project.github.io/motion_config.html#Options_Stream_Webcontrol). I plan to use this feature, but wanted to share this in case anyone else was curious about the security risks of turning `webcontrol_localhost` to `off`

> ALERT! Security Warning! This feature also means you have to pay attention to the following.
Anyone with access to the remote control port (http) can alter the values of options and save files anywhere on your server with the same privileges as the user running Motion. They can execute any command on your computer with the same privileges as the user running Motion. Anyone can access your control port if you have not either limited access to localhost or limited access using firewalls in the server. You should always have a router between a machine running Motion with remote control enabled and the Internet and make sure the Motion control port is not accessible from the outside. Also make sure to adjust the webcontrol_parms to the lowest level possible.
If you limit control port to localhost you still need to take care of any user logging into the server with any kind of terminal session.
Run Motion as a harmless user. DO NOT RUN AS ROOT!!

ccrisan, 2019-06-09:
@hamiltont indeed that setting is off by default in motionEye and should probably be left that way unless the unit runs in a local network where access is properly controlled.

kabadisha, 2019-06-09:
Good call @hamiltont and @ccrisan. I do run my cameras on a private local network, so no issue. I would be surprised if anyone was running their camera directly connected to the internet, but I guess statistics says that someone somewhere will...

pietje666, 2019-07-22:
> I have created a script that runs on my router and can detect if my phone is connected to the wifi network. If it is, it disables motion detection using the web control method above.
> 
> You can find the code here: https://gist.github.com/kabadisha/a87e50311d3ed4589f17a6e37c574a44

Hi i would like to use your script, do you have installation guide lines somewhere?
Because how do i run scripts on a router? Right now i have TP-link archer c7 with stock firmware. Do i need to flash DD-WRT firmware on it in order to run the script? And where should i put the script then? (was looking overhere https://wiki.dd-wrt.com/wiki/index.php/Script_Execution) But i do not see any logical place i would think it should run after any devices makes a connection?

pietje666, 2019-07-25:
@kabadisha not sure if you get notifications if i quote you so tagging you as well

kabadisha, 2019-07-25:
@pietje666 This is not a simple answer and will be quite a learning rabbit hole for you by the sounds of it, but I'll try and point you in the right direction :-)

In order for the script to work, you need the following things:
- Ability to put the script somewhere on non-volatile (so it sticks around between reboots) storage somewhere on your router and ability to make it executable with command `chmod +x`
- Ability to install the script in the router's crontab in order to schedule it's execution every few mins.
- A router who's firmware supports the `wl -i eth1 assoclist` or has something similar that you can change the script to use instead. I have no idea if DD-WRT supports that command, but there is a decent chance. You might want to ask someone on the DD-WRT forum. I use the Merlin firmware on an Asus router.

All of the above are probably not available on any stock firmware, so something like DD-WRT or Tomato is your best bet. The wiki and forums for those are the best place to find out how to install them and how to do the above once you have.

The above will also be fiddly to learn how to do, especially for someone who is not mega familiar with Unix and the command line, so Google and patience are your friends.

Hope that points you in the right direction :-)

pietje666, 2019-08-10:
@kabadisha thanks for the explanation. I got it working now using dd-wrt and cron job had to alter your script a bit since my router did not work with wl command had to use wl_atheros instead.

piachuPL, 2019-09-13:
I make python script to make ON/OFF motion detection by button i think that can help somebody 
https://github.com/piachuPL/Motionoffon/

rafik24, 2019-09-13:
piachuPL Many Thanks !

Rafik

 
 

https://github.com/piachuPL/Motionoffon/

divemasterjm, 2020-05-07:
i use hassOS on a virtualbox, can you please tell me where to find /data/etc/motion.conf ?

MYeager1967, 2020-07-23:
I know this is probably a long shot, but can a still picture be taken by webcontrol?

xraywinedrinker, 2020-09-11:
> There is a way - motioneye runs motion, which has a web control port (you can find the port number in the motion.conf configuration file, usually located in /etc/motioneye/motion.conf). I use it [...]

I would like to turn motion detection on/off via webcore when my location mode in smartthings is away. Can this command be sent via webcore and if so, do you know the proper web request format?

shooftie, 2020-10-04:
Is it just me or does the status of motion detection not update in the MotionEye UI? I understand that we are sidestepping MotionEye by fondling Motion directly but it would be helpful to see this represented in the MotionEye UI.

My assumption was that MotionEye was simply depicting the underlying Motion config but this would seem not to be the case.

MYeager1967, 2020-10-06:
There are many things that don't change in motionEye when you change them in motion via webcontrol. They get out of sync when you manipulate motion directly...

jbaragry, 2021-01-18:
I can start and pause detection through the API successfully but I notice notifications aren't sent when motion is started from the API rather than motioneye. (webhook notification)
I guess this is part of motioneye rather than motion. Anyone know how to notifications could be triggered through an API?

Dan1jel, 2023-02-03:
Just a hint for them like me who tried to find this solution and this took a while for me to get it to work, first, create the file inside `/data/etc/` , (not `/etc/xxx`) with `touch /data/etc/alarm_off_1` and insde that file `/data/etc/alarm_off_1` add the following below:

```
#!/usr/bin/bash
curl http://localhost:8080/0/detection/pause
```
and for `/data/etc/alarm_on_1` add the following below:

```
#!/usr/bin/bash
curl http://localhost:8080/0/detection/start
```

remember to change localhost (could work with IP adress) and the port (8080 in this example) and camera id (0 in this example) to your configuration.

closed 2026-11-01 by Marijn0

MichaIng, 2026-11-01:
An API and options buttons have been added with #3468, to toggle the  motion detection state via web UI, or manually via motionEye API.