motioneye-project/motioneye #803: Multiple devices with a central server

Homer-Rivera, 2018-02-23:
I have setup my centos 7 server running motioned with motion complied.  I have also setup my raspberry pi zero as a fast network camera using motioneyeos.  I have added my raspberry pi zero as a camera to the motioneye running on my hub (centos7 box).  The pi is configured as a fast network camera.

 I wanted to know if the central server/hub can monitor the streams for motion and create the movies local to the hub.  When I have added the remote camera I see no option to set a motion setting.  Is that possible to have the central server/hub do all the motion processing?

chrisf4lc0n, 2018-02-23:
Have you tried other camera types, not just fast network camera? I personally found the MotionEyeOS a bit to heavy on the little Pi 1 resources and created a simple stream with h264_v4l2_rtspserver, so all the motion detection happens on the server/hub side. There is however a small drawback to that, the framerate seems to be bit inconsistent, but that is all.

JamBalaya56562, 2026-11-07:
Closing as answered: a "Remote motionEye Camera" leaves motion detection to the motionEye on the Pi, so the hub has no motion settings for it. To have the hub do the detection and keep the movies, add the Pi's stream URL as a "Network Camera" instead; the hub's Motion then processes it, as suggested above.

closed 2026-11-07 by JamBalaya56562