motioneye-project/motioneye #804: motioneye camera noir with ir - white screen

kessero, 2018-02-24:
I have motioneye on raspberry pi 2 with Camera HD IR-CUT Noir OV5647 waveshare. When it's dark I have a white screen and no motion detection. I changed frames per seconds, but this doesn't help. Is there some trick to set up this camera?

chrisf4lc0n, 2018-02-24:
This camera is rather power hungry (especially the IRs), so check your PSU is powerful enough.

kessero, 2018-02-24:
> This camera is rather power hungry (especially the IRs), so check your PSU is powerful enough.
Raspistill works OK at night. My PSU i 3,1A should be enough.

chrisf4lc0n, 2018-02-24:
Does the raspistill command take a screenshot?

kessero, 2018-02-24:
Yes

chrisf4lc0n, 2018-02-24:
Set logging to debugging level and post some of the log files please.

EpicLPer, 2018-04-26:
Is it possible that you've mounted your camera in front of a window directly and the IR LEDs reflect the light too much and thus just make the screen white? Could be possible :)

kessero, 2018-04-26:
No camera was in dark room without any reflect. I think that problem is from PSU and from PIR detector. Now from about 10 days I remove PIR sensor and camera works OK but is testing most in light room so IR works with minimum power, sometimes in night too. Maybe PIR sensor is damage. Another theory is that when signal GSM/GPRS is low modem takes more power and PSU can't handle it.

jasaw, 2018-04-27:
@kessero You have GSM modem attached to the Pi? GSM modems usually have very high transient power consumption, so you have to make sure your power supply can handle GSM modem + IR LEDs + PIR detector + Pi itself with camera. Adding a few mF super-capacitor to your GSM modem supply may solve your problem.

JamBalaya56562, 2026-11-09:
As found here, this was most likely power: the IR LEDs, the PIR sensor and the GSM modem together were too much for the supply, and it worked after the PIR sensor was removed. Closing, as there was no news since 2018.

closed 2026-11-09 by JamBalaya56562