motioneye-project/motioneye #805: Video downloads

ADYoungberg, 2018-02-24:
Love this program! Running on a Raspberry Pi 3
Have a problem with downloading videos through the web interface VS accessing the same file on the HDD. Video downloaded through the web interface is locked on the 1st frame. Video on HDD plays correctly. I've attached the videos to show.  
Do I have a setting set up wrong?

Thanks again for the help!

[Video Files.zip](https://github.com/ccrisan/motioneye/files/1754237/Video.Files.zip)

ccrisan, 2018-02-26:
Are you saying that the downloaded file is different from the one present on HDD?

ADYoungberg, 2018-02-26:
It appears that way. I'll test it on a few other computers to ensure it's not a problem with the one computer.

ADYoungberg, 2018-02-26:
Found the problem... VLC media player doesn't like the MP4 files generated by MotionEye, but they play just fine with Windows Media and the Android video player. I'll just use those if I need to view something. Thanks!

chrisf4lc0n, 2018-03-05:
@ADYoungberg since your issue has been resolved could you close it, please?

closed 2018-03-05 by ADYoungberg