motioneye-project/motioneye #806: Problem with GDrive upload when MotionEye automatically start

VanLoon85, 2018-02-26:
Hi, to all, this is my first post.

I have a problem with Google Drive upload.

When MotionEye start automatically at startup the upload not working, if i try to stop the service and lauch meyectl and the upload working proptely (but in this mode the remote HTTP access not working).

At startup i have this situation:
root       351  2.2  2.5  63888 23776 ?        Ssl  12:26   0:16 /usr/bin/python /usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf
root      1275 52.8  9.6 302656 90896 ?        Sl   12:27   6:23 /usr/bin/motion -n -c /etc/motioneye/motion.conf -d 5
root      1350  0.0  1.3  32516 12612 ?        S    12:27   0:00 /usr/bin/python /usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf

When i stop the service and launch the comand meyectl i have this situation:
root      2648 49.1  9.8 374112 93112 pts/0    Sl+  12:40   0:17 /usr/bin/motion -n -c /etc/motioneye/motion.conf -d 5

What i doing wrong?

chrisf4lc0n, 2018-02-26:
It looks like the MotionEye service starts up before the network becomes available. What OS?

VanLoon85, 2018-02-26:
Raspbian 9.3

chrisf4lc0n, 2018-02-27:
Try this:
[https://unix.stackexchange.com/questions/126009/cause-a-script-to-execute-after-networking-has-started](https://unix.stackexchange.com/questions/126009/cause-a-script-to-execute-after-networking-has-started)

VanLoon85, 2018-03-01:
Unfortunately not working for me.

I noticed another thing, when i run the command meyectl startserver -d MotionEye don't load my motioneye config file in the folder /etc/motioneye.conf

   DEBUG: listing config dir /etc/motioneye...
   DEBUG: found camera with id 3
   DEBUG: found camera with id 1
   DEBUG: found camera with id 2
   DEBUG: reading camera config from /etc/motioneye/thread-1.conf...
   DEBUG: loading additional config structure for camera, without separators
   DEBUG: reading camera config from /etc/motioneye/thread-2.conf...
   DEBUG: reading camera config from /etc/motioneye/thread-3.conf...
   DEBUG: starting motion
   DEBUG: starting motion binary "/usr/bin/motion"

In this mode the GDrive upload working, but motioneye listening on default port, if i launch meyectl whit the argument -c and the position of config file ( /etc/motioneye/motioneye.conf) motioneye listening on the custom port, but the upload not working....

I have to check something in my motioneye.conf ?

JamBalaya56562, 2026-11-09:
The manual start without `-c` used a different configuration than the service, which makes the two runs hard to compare. The Google Drive upload has been reworked since 2018, and there was no news since, so closing. If uploads still fail when started as a service on a current version, please open a new issue with the motionEye log.

closed 2026-11-09 by JamBalaya56562