motioneye-project/motioneye #807: Issue with Pillow-5.0.0 when installing in Raspbian

mixpc, 2018-02-26:
Hello,

Brand new Raspbian install on Raspberry Pi 3 (2017-11-29-raspbian-stretch-lite.img).
After apt-get update and upgrade, Wiki instructions to install on Raspbian are followed. When issuing the command below, the error is displayed (note, Package: python Version: 2.7.13-2 State: installed)
Use of cached versions is due to reboot and new tries.

Could you please, share your views? Anyone had the same issue?

```
root@raspberrypi:~# pip install motioneye
Collecting motioneye
  Using cached motioneye-0.38.1.tar.gz
Collecting jinja2 (from motioneye)
  Using cached Jinja2-2.10-py2.py3-none-any.whl
Collecting pillow (from motioneye)
  Downloading Pillow-5.0.0.tar.gz (14.2MB)
    99% |████████████████████████████████| 14.2MB 1.5MB/s eta 0:00:01Exception:
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/pip/basecommand.py", line 215, in main
    status = self.run(options, args)
  File "/usr/lib/python2.7/dist-packages/pip/commands/install.py", line 353, in run
    wb.build(autobuilding=True)
  File "/usr/lib/python2.7/dist-packages/pip/wheel.py", line 749, in build
    self.requirement_set.prepare_files(self.finder)
  File "/usr/lib/python2.7/dist-packages/pip/req/req_set.py", line 380, in prepare_files
    ignore_dependencies=self.ignore_dependencies))
  File "/usr/lib/python2.7/dist-packages/pip/req/req_set.py", line 620, in _prepare_file
    session=self.session, hashes=hashes)
  File "/usr/lib/python2.7/dist-packages/pip/download.py", line 821, in unpack_url
    hashes=hashes
  File "/usr/lib/python2.7/dist-packages/pip/download.py", line 659, in unpack_http_url
    hashes)
  File "/usr/lib/python2.7/dist-packages/pip/download.py", line 882, in _download_http_url
    _download_url(resp, link, content_file, hashes)
  File "/usr/lib/python2.7/dist-packages/pip/download.py", line 603, in _download_url
    hashes.check_against_chunks(downloaded_chunks)
  File "/usr/lib/python2.7/dist-packages/pip/utils/hashes.py", line 46, in check_against_chunks
    for chunk in chunks:
  File "/usr/lib/python2.7/dist-packages/pip/download.py", line 571, in written_chunks
    for chunk in chunks:
  File "/usr/lib/python2.7/dist-packages/pip/utils/ui.py", line 139, in iter
    for x in it:
  File "/usr/lib/python2.7/dist-packages/pip/download.py", line 560, in resp_read
    decode_content=False):
  File "/usr/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/urllib3/response.py", line 432, in stream
    data = self.read(amt=amt, decode_content=decode_content)
  File "/usr/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/urllib3/response.py", line 380, in read
    data = self._fp.read(amt)
  File "/usr/share/python-wheels/CacheControl-0.11.7-py2.py3-none-any.whl/cachecontrol/filewrapper.py", line 63, in read
    self._close()
  File "/usr/share/python-wheels/CacheControl-0.11.7-py2.py3-none-any.whl/cachecontrol/filewrapper.py", line 50, in _close
    self.__callback(self.__buf.getvalue())
  File "/usr/share/python-wheels/CacheControl-0.11.7-py2.py3-none-any.whl/cachecontrol/controller.py", line 275, in cache_response
    self.serializer.dumps(request, response, body=body),
  File "/usr/share/python-wheels/CacheControl-0.11.7-py2.py3-none-any.whl/cachecontrol/serialize.py", line 87, in dumps
    ).encode("utf8"),
MemoryError
```

ccrisan, 2018-02-27:
I'm sorry but this is not a valid motionEye issue. It looks to me more like a `pillow` issue (on the Raspberry PI).

closed 2018-02-27 by ccrisan

mixpc, 2018-02-27:
No problem. Just for reference, I have followed the same steps in a Raspberry Pi 3 and install is successful so it might have to do with memory limits. In any case, may this be of use to anyone else running into the same issue.

atlex, 2018-10-23:
I have the same issue on my old Raspberry Pi A.
The solution for me is:
```
export PIP_NO_CACHE_DIR=false
pip install motioneye
```