motioneye-project/motioneye #808: Add option to create blank .nomedia file for Google Drive uploads

JSkier21, 2018-02-26:
I would like to have .nomedia added to the Google Cloud upon upload (as an option) please. Ideally this would work for subfolders on or off, however it would be most useful when creating with new sub-folders. 

It's easy enough to create the blank file when just using the directory, but it becomes cumbersome when motioneye creates new directories. Google Photos picks up the pics and videos and adds it to the Photo app, which I don't want.

Kcarpenter528, 2018-02-28:
Is there a reason why you can't turn off the Google Photos/Drive integration? Google Photos App -> Settings -> Toggle "Google Drive - Show Google Drive Photos and Videos in your Photos Library"

Or do you want certain drive folders to show up in Photos?

JSkier21, 2018-03-01:
Google Drive is off, it's still showing the folder to users I have shared with (on both the computer and phone it has been off). Ideally I would like to control this per directory.

chrisf4lc0n, 2018-03-04:
What if you added a script creating an empty .nomedia file on motion detection?
Something along them lines:
```
#!/bin/bash
d=$(date +%Y-%m-%d)
cd /path/to/the/folder/you/store/your/media/"$d"
touch .nomedia
```

JSkier21, 2018-03-12:
I'll give it a shot; is this what gets uploaded to Google Drive?

chrisf4lc0n, 2018-03-13:
Everything within the folder you set it up for will get uploaded to GDrive.

JamBalaya56562, 2026-11-09:
A script on motion detection that creates the `.nomedia` file was suggested as a workaround, and there was no news since 2018. Closing, as this is rather a Google Photos setting than something motionEye should create in every folder.

closed 2026-11-09 by JamBalaya56562