motioneye-project/motioneye #809: Raspbian Install Issue

smjohnson85, 2018-02-28:
I kept getting an error on startup until I reinstalled with pip2:
`systemctl status motioneye
● motioneye.service - motionEye Server
   Loaded: loaded (/etc/systemd/system/motioneye.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Wed 2018-02-28 01:42:47 PST; 12min ago
  Process: 9946 ExecStart=/usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf (code=exited, status=203/EXEC)
 Main PID: 9946 (code=exited, status=203/EXEC)

Feb 28 01:42:47 raspberrypi systemd[1]: Started motionEye Server.
Feb 28 01:42:47 raspberrypi systemd[1]: motioneye.service: Main process exited, code=exited, status=203/EXEC
Feb 28 01:42:47 raspberrypi systemd[1]: motioneye.service: Unit entered failed state.
Feb 28 01:42:47 raspberrypi systemd[1]: motioneye.service: Failed with result 'exit-code'.`

This is on latest Raspbian stretch. Would be helpful if wiki step 5 was updated to reflect need to use pip2 command.
https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian

ccrisan, 2018-02-28:
Is python 3 already the default on Raspbian?

chrisf4lc0n, 2018-03-01:
Issue command 'python' and post the results.

smjohnson85, 2018-03-03:
Python 2.7.13 but it appears that both pip and pip2 are installed by default.

davesdere, 2018-06-05:
Wow... I tried pip, pip3 but I didn't even know pip2 existed... I assumed that pip was pip2. Now I just tried, but it still doesn't work. Anybody got further on this?

ccrisan, 2018-06-09:
@davesdere we have some Docker stuff that you can find in the [extra](https://github.com/ccrisan/motioneye/tree/master/extra) folder of the project. See if that works for you.

davesdere, 2018-06-09:
Thanks @ccrisan ! I will check it out. While I'm here...
`pip2.7 install motioneye`
Solved it for me.
BTW, you did an amazing job with this UI. It's stable, simple and easy to configure. You are a pro!

closed 2022-03-19 by MichaIng