motioneye-project/motioneye #810: Motion Notification - Script does not execute.

Gerry4321, 2018-03-01:
Hello,
I have:
motionEye Version	0.38.1
Motion Version	4.0.1
OS Version	Raspbian 8.0

"Run a command" On
"Command" /home/pi/Meye_GF/MotionEYE_GF.sh

"MotionEYE_GF.sh" =
#!/bin/bash
sudo echo "Motion Detected"

Permissions = 
drwxr-xr-x  2 motion motion     4096 Feb 28 19:11 Meye_GF
-rwxr-xr-x 1 motion motion 86 Feb 28 19:11 MotionEYE_GF.sh

motion.log shows "Motion Detected"

No errors in syslog relevant to the script.

Script does nothing - can you please help ?

chrisf4lc0n, 2018-03-03:
Hi,
I could be wrong on this one, but if the script is executed in the background not as a live user it will not print "Motion Detection" on the screen. Try:
`echo "Motion Detection" > motion.detection.file`
and see if the file gets created

ccrisan, 2018-03-03:
@Gerry4321 your motioneye log file should give you more details on why the script execution did not succeed.

Gerry4321, 2018-03-03:
@chrisf4lc0n  Thank you, but no difference.
@ccrisan Thank you, but I have no motioneye.log just a motion.log ?

ccrisan, 2018-03-04:
@Gerry4321 on a systemd-based OS, you can use the following command:

    journalctl -u motioneye > motioneye.log

Gerry4321, 2018-03-04:
journalctl -u motioneye > motioneye.log
No journal files were found.

ccrisan, 2018-03-04:
@Gerry4321 how so you run motionEye?

Gerry4321, 2018-03-04:
I have an init script, to run at startup & start the motioneye server.

chrisf4lc0n, 2018-03-04:
Not systemd based OS then...
The motioneye logs should be in the system.log, correct me if I am wrong Calin.

ccrisan, 2018-03-04:
@chrisf4lc0n on sysvinit-based OSes, if the user installed the recommended init script, the log goes to `/var/log/motioneye.log`. However there are a thousand other possibilities of running motionEye, so I can't say for sure where the log is.

@Gerry4321 you're not giving us too much information. We can't help you unless you're telling us what init script you use and how exactly is the `meyectl` command effectively run.

Gerry4321, 2018-03-04:
Rightly or wrongly this is the guide & method I followed -
http://www.instructables.com/id/Build-Your-Own-Surveillance-Camera-With-Raspberry-/

ccrisan, 2018-03-05:
@Gerry4321 so you are on systemd. What does `sudo journalctl -u motioneye` say?

Gerry4321, 2018-03-05:
-- Logs begin at Sun 2018-03-04 07:35:34 GMT, end at Mon 2018-03-05 12:59:20 GMT. --
Mar 04 07:35:36 raspberrypi systemd[1]: Starting motionEye Server...
Mar 04 07:35:36 raspberrypi systemd[1]: Started motionEye Server.
Mar 04 07:35:50 raspberrypi meyectl[378]: INFO: hello! this is motionEye server 0.38.1
Mar 04 07:35:52 raspberrypi meyectl[378]: INFO: cleanup started
Mar 04 07:35:52 raspberrypi meyectl[378]: INFO: wsswitch started
Mar 04 07:35:52 raspberrypi meyectl[378]: INFO: tasks started
Mar 04 07:35:52 raspberrypi meyectl[378]: INFO: mjpg client garbage collector started
Mar 04 07:35:52 raspberrypi meyectl[378]: INFO: server started
Mar 04 14:25:05 raspberrypi meyectl[378]: WARNING: 403 GET /login/?_=1520173536695&_username=_&_signature=a893fe3bb9887e2c9222a51edb50378fa2176bdc (192.168.2.19) 4.42ms
Mar 04 14:25:22 raspberrypi motion[5523]: [0:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
Mar 04 14:25:22 raspberrypi motion[5523]: [0:motion] [ALR] [ALL] conf_cmdparse: Unknown config option "webcontrol_parms"
Mar 04 14:25:22 raspberrypi motion[5523]: [0:motion] [WRN] [ALL] thread config option deprecated use camera
Mar 04 14:25:22 raspberrypi motion[5523]: [0:motion] [NTC] [ALL] config_camera: Processing camera config file thread-1.conf
Mar 04 14:25:22 raspberrypi motion[5523]: [0:motion] [NTC] [ALL] motion_startup: Motion 4.0.1 Started
Mar 04 14:25:22 raspberrypi motion[5523]: [0:motion] [NTC] [ALL] motion_startup: Logging to syslog
Mar 04 14:25:22 raspberrypi motion[5523]: [0:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
Mar 04 14:25:22 raspberrypi motion[5523]: [0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
Mar 04 14:25:22 raspberrypi motion[5523]: [1:ml1] [WRN] [VID] v4l2_select_input: Device doesn't support VIDIOC_G_STD
Mar 04 14:26:18 raspberrypi sudo[5569]: root : TTY=unknown ; PWD=/etc/motioneye ; USER=root ; COMMAND=/bin/echo Motion Detected_gf
Mar 04 14:26:18 raspberrypi sudo[5569]: pam_unix(sudo:session): session opened for user root by (uid=0)
Mar 04 14:26:18 raspberrypi sudo[5569]: pam_unix(sudo:session): session closed for user root
Mar 04 14:27:58 raspberrypi sudo[5638]: root : TTY=unknown ; PWD=/etc/motioneye ; USER=root ; COMMAND=/bin/echo Motion Detected_gf
Mar 04 14:27:58 raspberrypi sudo[5638]: pam_unix(sudo:session): session opened for user root by (uid=0)
lines 1-24

ccrisan, 2018-03-05:
Great. No do that again after running your command and redirect the output to a file. Then attach the log.

Gerry4321, 2018-03-05:
[somefile2.txt](https://github.com/ccrisan/motioneye/files/1780942/somefile2.txt)

ccrisan, 2018-03-05:
@Gerry4321 I'm seeing some strange lines in your log:

    Mar 04 18:37:01 raspberrypi sudo[9881]: root : TTY=unknown ; PWD=/etc/motioneye ; USER=root ; COMMAND=/bin/echo Motion Detected_gf
    Mar 04 18:37:01 raspberrypi sudo[9881]: pam_unix(sudo:session): session opened for user root by (uid=0)
    Mar 04 18:37:01 raspberrypi sudo[9881]: pam_unix(sudo:session): session closed for user root
    Mar 04 18:37:01 raspberrypi sudo[9890]: root : TTY=unknown ; PWD=/etc/motioneye ; USER=root ; COMMAND=/bin/echo 2018-03-04---18:37:01 Motion Detected

`COMMAND=/bin/echo Motion Detected_gf` - what's up with that command?

chrisf4lc0n, 2018-03-05:
It looks to me like the script gets executed.

Gerry4321, 2018-03-05:
@ccrisan "what's up with that command?" Where does it get output / executed ? Where is the motioneye.log or is their one?

chrisf4lc0n, 2018-03-05:
@Gerry4321 Sorry I do not get it where would you like to have the output of that echo? If you want a log you have to set it up to log into a file by what has already been mentioned:
`journalctl -u motioneye > motioneye.log`

Gerry4321, 2018-03-05:
@chrisf4lc0n Thank you, I understand now. As to "journalctl -u motioneye > motioneye.log" I tried this allready & got "No journal files were found." - may you explain, please ?

chrisf4lc0n, 2018-03-05:
I take it you have tried it as sudo and while motioneye was running?

Gerry4321, 2018-03-05:
I have now... All OK. Issue closed.
Thank you both, very much.
G.

closed 2018-03-05 by Gerry4321