motioneye-project/motioneye #811: issue with adding network device

Olegone23, 2018-03-01:
Hi Cris, hope u are well!

Could you pls. help with my issue (just trying to add  a network cam):
I badly need to use old version of linux and motion too.

my soft versions are:

motionEye Version 	0.38
Motion Version 	3.2.12

motionlog:
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/tornado/web.py", line 1046, in _stack_context_handle_exception
    raise_exc_info((type, value, traceback))
  File "/usr/lib/python2.7/dist-packages/tornado/web.py", line 1164, in wrapper
    return method(self, *args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/motioneye-0.38-py2.7.egg/motioneye/handlers.py", line 275, in get
    self.list()
  File "/usr/local/lib/python2.7/dist-packages/motioneye-0.38-py2.7.egg/motioneye/handlers.py", line 205, in wrapper
    return func(self, *args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/motioneye-0.38-py2.7.egg/motioneye/handlers.py", line 623, in list
    callback=on_response)
  File "/usr/local/lib/python2.7/dist-packages/motioneye-0.38-py2.7.egg/motioneye/utils.py", line 441, in test_mjpeg_url
    do_request(on_response)
  File "/usr/local/lib/python2.7/dist-packages/motioneye-0.38-py2.7.egg/motioneye/utils.py", line 398, in do_request
    header_callback=on_header, validate_cert=settings.VALIDATE_CERTS)
TypeError: __init__() got an unexpected keyword argument 'auth_mode'
2018-03-01 19:38:42: [motioneye]    ERROR: 500 GET /config/list/?_=1519922322494&scheme=http&host=91.195.136.170&port=&path=%2Fmjpg%2Fvideo.mjpg&username=&password=&proto=netcam&_username=admin&_signature=d033b567324385f14f6af461ab7c3856ba53f99c (127.0.0.1) 45.37ms

Kind Regards,
Oleg

ccrisan, 2018-03-03:
What is the version of tornado you're using? The following line should tell you that:

    python -c 'import tornado; print(tornado.version)'

Olegone23, 2018-03-03:
Seeems 2.3

Olegone23, 2018-03-03:
What I've got is: filr utils.py-> string 395 -> auth_ mode parameter (by default is basic),
I've tired to remove it from HTTPreguest string and camera was added (appeared in UI), but with black screen, seems something wrong with motion, may be, but recorded movie is good quality, the problem only with black screen

ccrisan, 2018-03-04:
@Olegone23 your tornado version is painfully old. Please upgrade your tornado python package.

closed 2018-03-04 by ccrisan