motioneye-project/motioneye #812: No record

Olegone23, 2018-03-05:
Hi Cris,

Updated Tornado from 2.7  to 3.1 (can't up higher then that, GCC dependencies)
got new issue (no records in file, I can see the file with the first frame only):
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/tornado-3.1-py2.7.egg/tornado/web.py", line 1115, in _stack_context_handle_exception
    raise_exc_info((type, value, traceback))
  File "/usr/local/lib/python2.7/dist-packages/tornado-3.1-py2.7.egg/tornado/web.py", line 1298, in wrapper
    result = method(self, *args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/motioneye-0.38-py2.7.egg/motioneye/handlers.py", line 970, in get
    self.current(camera_id)
  File "/usr/local/lib/python2.7/dist-packages/motioneye-0.38-py2.7.egg/motioneye/handlers.py", line 205, in wrapper
    return func(self, *args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/motioneye-0.38-py2.7.egg/motioneye/handlers.py", line 1033, in current
    camera_id=camera_id, retry=retry + 1)
TypeError: add_timeout() got an unexpected keyword argument 'camera_id'
   ERROR: 500 GET /picture/2/current/?_=1520243395746&_username=admin&_signature=a90f5ec9c18718b2dc4fb7ce645e8e1c40cbf295 (127.0.0.1) 0.79ms

ccrisan, 2018-03-05:
@Olegone23 unfortunately motionEye does not work with tornado that old. I know it used to work with tornado 3.1+ but things have changed a bit (and I also have to update the minimum required version in `setup.py`). This being said, please upgrade to tornado 4.x or use an older version of motionEye.

closed 2018-03-05 by ccrisan

Olegone23, 2018-03-05:
Thank you! What is a best choice of MotionEye for tornado 3.1?

ccrisan, 2018-03-05:
No idea, you'll have to test them one by one until you get one that works.

Olegone23, 2018-03-05:
thank u anyway for kind support!