motioneye-project/motioneye #816: Raspbian Stretch Install - Step 6 source for copy not found

jsavage, 2018-03-08:
Hi,
When installing on a virgin but updated and upgraded i(today) install of Raspbian Stretch on a Pi V2 I am seeing the following error at Step 6:

```
sudo cp /usr/local/share/motioneye/extra/motioneye.conf.sample /etc/motioneye/motioneye.conf
cp: cannot stat '/usr/local/share/motioneye/extra/motioneye.conf.sample': No such file or directory
```

Step 6 reads as follows::
```
Prepare the configuration directory:
 mkdir -p /etc/motioneye
 cp /usr/local/share/motioneye/extra/motioneye.conf.sample /etc/motioneye/motioneye.conf
```

The previous step is  pip install motioneye
This completed seemingly correctly - the output ending with: 
```
Successfully built motioneye pillow pycurl tornado MarkupSafe
Installing collected packages: MarkupSafe, jinja2, pillow, pycurl, backports-abc, futures, six, singledispatch, tornado, motioneye
Successfully installed MarkupSafe-1.0 backports-abc-0.5 futures-3.2.0 jinja2-2.10 motioneye-0.38.1 pillow-5.0.0 pycurl-7.43.0.1 singledispatch-3.4.0.3 six-1.11.0 tornado-5.0
```

I have seen a similar issue a few days ago suggesting use of pip2.  Will give that a go for now.

James

chrisf4lc0n, 2018-03-08:
`sudo nano /etc/motioneye/motioneye.conf`

and paste:

```
# path to the configuration directory (must be writable by motionEye)
conf_path /etc/motioneye

# path to the directory where pid files go (must be writable by motionEye)
run_path /var/run

# path to the directory where log files go (must be writable by motionEye)
log_path /var/log

# default output path for media files (must be writable by motionEye)
media_path /var/lib/motioneye

# the log level (use quiet, error, warning, info or debug)
log_level info

# the IP address to listen on
# (0.0.0.0 for all interfaces, 127.0.0.1 for localhost)
listen 0.0.0.0

# the TCP port to listen on
port 8765

# path to the motion binary to use (automatically detected if commented)
#motion_binary /usr/bin/motion

# whether motion HTTP control interface listens on
# localhost or on all interfaces
motion_control_localhost true

# the TCP port that motion HTTP control interface listens on
motion_control_port 7999

# interval in seconds at which motionEye checks if motion is running
motion_check_interval 10

# whether to restart the motion daemon when an error occurs while communicating with it
motion_restart_on_errors false

# interval in seconds at which motionEye checks the SMB mounts
mount_check_interval 300

# interval in seconds at which the janitor is called
# to remove old pictures and movies
cleanup_interval 43200

# timeout in seconds to wait for response from a remote motionEye server
remote_request_timeout 10

# timeout in seconds to wait for mjpg data from the motion daemon
mjpg_client_timeout 10

# timeout in seconds after which an idle mjpg client is removed
# (set to 0 to disable)
mjpg_client_idle_timeout 10

# enable SMB shares (requires motionEye to run as root) 
smb_shares false

# the directory where the SMB mount points will be created
smb_mount_root /media

# path to the wpa_supplicant.conf file
# (enable this to configure wifi settings from the UI)
#wpa_supplicant_conf /etc/wpa_supplicant.conf

# path to the localtime file
# (enable this to configure the system time zone from the UI)
#local_time_file /etc/localtime

# enables shutdown and rebooting after changing system settings
# (such as wifi settings or time zone)
enable_reboot false

# timeout in seconds to use when talking to the SMTP server
smtp_timeout 60

# timeout in seconds to wait for media files list
list_media_timeout 120

# timeout in seconds to wait for media files list, when sending emails
list_media_timeout_email 10

# timeout in seconds to wait for zip file creation
zip_timeout 500

# timeout in seconds to wait for timelapse creation
timelapse_timeout 500

# enable adding and removing cameras from UI
add_remove_cameras true

# enables HTTP basic authentication scheme (in addition to, not instead of the signature mechanism)
http_basic_auth false
```

ccrisan, 2018-03-08:
@jsavage I just ran myself every command on that article, one by one, on a freshly installed and up-to-date Raspbian. Everything worked smoothly and I got no error. I admit I used Raspbian lite. Did you mess around with the python version? Can you reproduce your problem?

jsavage, 2018-03-08:
It was the full version of stretch that I used.  I even checked the hash.  Had a minor glitch on first boot but boots in fine since.  Apt update Dist-upgrad. Added Bluetooth adaptor and wifi adaptors. SSH, VNC, 1-wire enabled. very odd.

Assume above workaround will suffice to workaround Step 6 but Step 8 is also failing as the source material is not where it should be in the share/motioneye folder.

At what step should this folder have been created.  Perhaps I can re-run that part?

James

jsavage, 2018-03-09:
Ah!   I think I have identified the issue...
I was running pip without sudo.
the output had been as follows:
```
pip install motioneye
Collecting motioneye
Collecting jinja2 (from motioneye)
  Using cached Jinja2-2.10-py2.py3-none-any.whl
Collecting tornado>=3.1 (from motioneye)
Collecting pillow (from motioneye)
Collecting pycurl (from motioneye)
Collecting MarkupSafe>=0.23 (from jinja2->motioneye)
Collecting singledispatch (from tornado>=3.1->motioneye)
  Using cached singledispatch-3.4.0.3-py2.py3-none-any.whl
Collecting futures (from tornado>=3.1->motioneye)
  Using cached futures-3.2.0-py2-none-any.whl
Collecting backports-abc>=0.4 (from tornado>=3.1->motioneye)
  Using cached backports_abc-0.5-py2.py3-none-any.whl
Collecting six (from singledispatch->tornado>=3.1->motioneye)
  Using cached six-1.11.0-py2.py3-none-any.whl
Installing collected packages: MarkupSafe, jinja2, six, singledispatch, futures, backports-abc, tornado, pillow, pycurl, motioneye
Successfully installed MarkupSafe-1.0 backports-abc-0.5 futures-3.2.0 jinja2-2.10 motioneye-0.38.1 pillow-5.0.0 pycurl-7.43.0.1 singledispatch-3.4.0.3 six-1.11.0 tornado-5.0
```
I have just tried executing this with sudo with the following result:
```
$ sudo pip install motioneye
Collecting motioneye
  Downloading motioneye-0.38.1.tar.gz (756kB)
    100% |████████████████████████████████| 757kB 208kB/s 
Requirement already satisfied: jinja2 in /usr/lib/python2.7/dist-packages (from motioneye)
Requirement already satisfied: pillow in /usr/lib/python2.7/dist-packages (from motioneye)
Collecting pycurl (from motioneye)
  Downloading pycurl-7.43.0.1.tar.gz (195kB)
    100% |████████████████████████████████| 204kB 688kB/s 
Collecting tornado>=3.1 (from motioneye)
  Downloading tornado-5.0.tar.gz (504kB)
    100% |████████████████████████████████| 512kB 304kB/s 
Requirement already satisfied: MarkupSafe in /usr/lib/python2.7/dist-packages (from jinja2->motioneye)
Collecting backports_abc>=0.4 (from tornado>=3.1->motioneye)
  Downloading backports_abc-0.5-py2.py3-none-any.whl
Collecting futures (from tornado>=3.1->motioneye)
  Downloading futures-3.2.0-py2-none-any.whl
Collecting singledispatch (from tornado>=3.1->motioneye)
  Downloading singledispatch-3.4.0.3-py2.py3-none-any.whl
Requirement already satisfied: six in /usr/lib/python2.7/dist-packages (from singledispatch->tornado>=3.1->motioneye)
Building wheels for collected packages: motioneye, pycurl, tornado
  Running setup.py bdist_wheel for motioneye ... done
  Stored in directory: /root/.cache/pip/wheels/97/97/ac/8abf7d495662b36e25492749b355ae84455c3ff8bfcd2212db
  Running setup.py bdist_wheel for pycurl ... done
  Stored in directory: /root/.cache/pip/wheels/a5/5b/c8/f80900b09b49815e1f90dbae2f57e49b3f4c61071db40fb238
  Running setup.py bdist_wheel for tornado ... done
  Stored in directory: /root/.cache/pip/wheels/ec/da/45/505e8bf2dc544b21aeeee8705da32a2b27600f541dca4a3c8f
Successfully built motioneye pycurl tornado
Installing collected packages: pycurl, backports-abc, futures, singledispatch, tornado, motioneye
Successfully installed backports-abc-0.5 futures-3.2.0 motioneye-0.38.1 pycurl-7.43.0.1 singledispatch-3.4.0.3 tornado-5.0
```
Step 6 now works as expected.

Now I am wary of over-use of sudo so occasionally err on the side of caution and rely on inadequate permissions becoming obvious.  On this occasion, no warnings or errors were issued so that strategy failed.

I am not sure what the solution is here. Those more familiar with pip than t would not have fallen in this hole but I do wonder whether it might be worth checking permissions are adequate in the pip install script.

ccrisan - thanks for checking the script - it gave me added confidence to persevere. I have motioneyeos running successfully in a separate device - for which many thanks too.

closed 2018-03-13 by ccrisan