motioneye-project/motioneye #817: Unable to add a Remote MotionEye Camera

jsavage, 2018-03-09:
[image]
Here I am trying to add a remote MotionEye camera (motioneyeos) that is running on my LAN on 192,168.1.129  .  I access this motioneye device on port 80.

The motion eye I am configuring this on is a new install on Raspbian Stretch.  It already has one Raspi camera working. The remote camera is running MotionEyeos.

However, after entering the password, this dialog is not allowing me to select the camera.  
I am not sure whether the credentials are even being checked.  The credentials I am using are the admin ones that work from a web browser.  The screenshot shows URL 192.168.1.129/cams/  .  I am including 'cams' as suggested by the red '!' tool tip  but I have also tried using 192.168.1.129 but this also shows a red '!' icon too.

Questions: 
1. is there something different that I need to enable in MotioneyeOS to enable a camera to be accessed remotely?
2. How can I troubleshoot this?

chrisf4lc0n, 2018-03-09:
The URL does not look right.
1. It has to be either http:// or rtsp://
2. I you are connecting to another motionEye device you will be using the URL from that device on the port it is running on, not the 80. For example http://192.168.0.16:8765/

jsavage, 2018-03-10:
Thanks - http:// was all it needed.  But port 80 was correct possibly because I set it at that when originally configuring MotionEyeOS

closed 2018-03-10 by ccrisan