motioneye-project/motioneye #818: External storage problem

pcbjonnes, 2018-03-09:
I have tried using both ftp and dropbox for external storage. I enter the settings and press the test button and it seems to work. The problem is that nothing is being uploaded to the services. Does anyone have any suggestions to resolve this?

ccrisan, 2018-03-09:
1. What motionEye version? what motion version? what OS? what architecture/board/hardware?
2. Do you actually have media files locally?
3. Do you see any suspicious errors in your motioneye log file?

pcbjonnes, 2018-03-09:
1.
Rasberry Pi 3 model B
motionEye Version: 0.38.1
Motion Version: 4.1
OS Version: Raspbian 9.3

2.
Yes. Both movies and pictures are stored locally in /var/lib/motioneye/Camera1/

3.
/var/log/motion.log
The complete logfile:
29111832:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
[29111832:motion] [ALR] [ALL] conf_cmdparse: Deprecated config option "thread" since after version 3.4.1:
[29111832:motion] [ALR] [ALL] conf_cmdparse: The "thread" option has been replaced by the "camera" option.
[29111832:motion] [NTC] [ALL] config_camera: Processing camera config file thread-1.conf
[29111832:motion] [NTC] [ALL] motion_startup: Motion 4.1 Started
[29111832:motion] [NTC] [ALL] motion_startup: Logging to syslog
[29111832:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
[29111832:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
[0:motion] [ERR] [ALL] main: Thread 1 - Watchdog timeout, trying to do a graceful restart
[0:motion] [ERR] [ALL] main: Thread 1 - Watchdog timeout, did NOT restart graceful, killing it!
[0:motion] [ERR] [ALL] main: cleaning Thread 1

pcbjonnes, 2018-03-09:
Well... for some reason the cleanup just started working right now. Thank you anyway!

pcbjonnes, 2018-03-16:
I do not know if this makes any sense to you @ccrisan , but I've discovered something about this.
I reinstalled MotionEye and set up transfer to FTP. To my surprise, it worked. Then I proceeded to configure the system, including putting password on admin. Afterwards, the upload did not work anymore. I analyzed what I had done and tried to set admin password back to nothing again. Then MotionEye started to transfer to FTP again.
I have tried this several times now and the same happens every time.
Anyway - thanks.

ccrisan, 2018-03-16:
@pcbjonnes it actually makes sense in the way that events are forwarded from the motion daemon to motionEye via internal web hooks and the actual HTTP request is handled by [relayevent.sh](https://github.com/ccrisan/motioneye/blob/master/motioneye/scripts/relayevent.sh). It uses your admin password to authenticate to motionEye and it may be failing due to some special characters you have in your admin password.

What you could do is try with simpler admin passwords and see if there's a special character that creates problems. I'll make sure to escape it afterwards. Investigation motionEye's log during all this (and looking for 403s) could help.

e36Alex, 2018-06-15:
Hello!

I got a similar issue. For months i am used to upload the media files to my nas and to my cloud storage box with ftp - everything works fine until a few weeks.

since a few weeks files are just stored on my nas and not uploaded to the ftp.

i even tried to changed my admin assword to a simple one, but in my case it didnt change anything.

the service test of the ftp was successful.

here the content of the motion.log:
`[0:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
[0:motion] [WRN] [ALL] thread config option deprecated use camera
[0:motion] [NTC] [ALL] config_camera: Processing camera config file thread-1.conf
[0:motion] [ALR] [ALL] conf_cmdparse: Unknown config option "webcontrol_parms"
[0:motion] [NTC] [ALL] motion_startup: Motion 4.0 Started
[0:motion] [NTC] [ALL] motion_startup: Logging to syslog
[0:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
[0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
[1:ml1] [ALR] [NET] netcam_start: Network Camera thread starting... for url (rtsp://192.168.178.181:554/11)
[1:nc1] [ALR] [NET] netcam_handler_loop: Camera handler thread [3] started
curl: (3) Illegal characters found in URL
curl: (3) Illegal characters found in URL
curl: (3) Illegal characters found in URL
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed

  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100     2    0     0  100     2      0    446 --:--:-- --:--:-- --:--:--   500
curl: (3) Illegal characters found in URL
curl: (3) Illegal characters found in URL
curl: (3) Illegal characters found in URL
curl: (3) Illegal characters found in URL
curl: (3) Illegal characters found in URL
curl: (3) Illegal characters found in URL
curl: (3) Illegal characters found in URL
curl: (3) Illegal characters found in URL
curl: (3) Illegal characters found in URL
curl: (3) Illegal characters found in URL
curl: (3) Illegal characters found in URL
curl: (3) Illegal characters found in URL
curl: (3) Illegal characters found in URL
curl: (3) Illegal characters found in URL
curl: (3) Illegal characters found in URL
curl: (3) Illegal characters found in URL
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed

  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100     3    0     0  100     3      0    171 --:--:-- --:--:-- --:--:--   750
curl: (3) Illegal characters found in URL
curl: (3) Illegal characters found in URL
curl: (3) Illegal characters found in URL
curl: (3) Illegal characters found in URL
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed

  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100     2    0     0  100     2      0    176 --:--:-- --:--:-- --:--:--   181
curl: (3) Illegal characters found in URL
curl: (3) Illegal characters found in URL
curl: (3) Illegal characters found in URL
curl: (3) Illegal characters found in URL
curl: (3) Illegal characters found in URL
curl: (3) Illegal characters found in URL
curl: (3) Illegal characters found in URL
curl: (3) Illegal characters found in URL
curl: (3) Illegal characters found in URL
curl: (3) Illegal characters found in URL
curl: (3) Illegal characters found in URL
curl: (3) Illegal characters found in URL
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed

  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0curl: (3) Illegal characters found in URL

100     3    0     0  100     3      0    606 --:--:-- --:--:-- --:--:--   750
`

e36Alex, 2018-06-15:
Hello!

I got a similar issue. For months i am used to upload the media files to my nas and to my cloud storage box with ftp - everything works fine until a few weeks.

since a few weeks files are just stored on my nas and not uploaded to the ftp.

i even tried to changed my admin assword to a simple one, but in my case it didnt change anything.

the service test of the ftp was successful.

here the content of the motion.log:
`[0:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
[0:motion] [WRN] [ALL] thread config option deprecated use camera
[0:motion] [NTC] [ALL] config_camera: Processing camera config file thread-1.conf
[0:motion] [ALR] [ALL] conf_cmdparse: Unknown config option "webcontrol_parms"
[0:motion] [NTC] [ALL] motion_startup: Motion 4.0 Started
[0:motion] [NTC] [ALL] motion_startup: Logging to syslog
[0:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
[0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
[1:ml1] [ALR] [NET] netcam_start: Network Camera thread starting... for url (rtsp://192.168.178.181:554/11)
[1:nc1] [ALR] [NET] netcam_handler_loop: Camera handler thread [3] started
curl: (3) Illegal characters found in URL
curl: (3) Illegal characters found in URL
curl: (3) Illegal characters found in URL
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed

  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100     2    0     0  100     2      0    446 --:--:-- --:--:-- --:--:--   500
curl: (3) Illegal characters found in URL
curl: (3) Illegal characters found in URL
curl: (3) Illegal characters found in URL
curl: (3) Illegal characters found in URL
curl: (3) Illegal characters found in URL
curl: (3) Illegal characters found in URL
curl: (3) Illegal characters found in URL
curl: (3) Illegal characters found in URL
curl: (3) Illegal characters found in URL
curl: (3) Illegal characters found in URL
curl: (3) Illegal characters found in URL
curl: (3) Illegal characters found in URL
curl: (3) Illegal characters found in URL
curl: (3) Illegal characters found in URL
curl: (3) Illegal characters found in URL
curl: (3) Illegal characters found in URL
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed

  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100     3    0     0  100     3      0    171 --:--:-- --:--:-- --:--:--   750
curl: (3) Illegal characters found in URL
curl: (3) Illegal characters found in URL
curl: (3) Illegal characters found in URL
curl: (3) Illegal characters found in URL
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed

  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100     2    0     0  100     2      0    176 --:--:-- --:--:-- --:--:--   181
curl: (3) Illegal characters found in URL
curl: (3) Illegal characters found in URL
curl: (3) Illegal characters found in URL
curl: (3) Illegal characters found in URL
curl: (3) Illegal characters found in URL
curl: (3) Illegal characters found in URL
curl: (3) Illegal characters found in URL
curl: (3) Illegal characters found in URL
curl: (3) Illegal characters found in URL
curl: (3) Illegal characters found in URL
curl: (3) Illegal characters found in URL
curl: (3) Illegal characters found in URL
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed

  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0curl: (3) Illegal characters found in URL

100     3    0     0  100     3      0    606 --:--:-- --:--:-- --:--:--   750
`
System-Info:
motionEye Version | 0.39.1
-- | --
Motion Version | 4.0
OS Version | Debian 9.4

JamBalaya56562, 2026-11-07:
Closing, as the cause found above is gone: Motion's events used to reach motionEye with the admin password, which failed with some passwords and stopped the uploads. Since 0.44.0 (#3332) `relayevent.sh` authenticates with a separate secret instead. If uploads still do not start with a current release while the service test passes, please open a new issue with `motioneye.log`.

closed 2026-11-07 by JamBalaya56562