motioneye-project/motioneye #820: Cannot find mass storage

jpihtila, 2018-03-10:
Raspberry pi 3 with motion eye os.
I cant find at attached usb hard drive. It's not in drop down menu. 
I appreciate any help in advance.
[boot.log](https://github.com/ccrisan/motioneye/files/1799523/boot.log)
[dmesg.log](https://github.com/ccrisan/motioneye/files/1799524/dmesg.log)
[messages.log](https://github.com/ccrisan/motioneye/files/1799525/messages.log)
[motion.log](https://github.com/ccrisan/motioneye/files/1799526/motion.log)
[motioneye.log](https://github.com/ccrisan/motioneye/files/1799527/motioneye.log)

ccrisan, 2018-03-10:
Given the following lines, I suspect you have more tha one USB storage device attached, namely `sda` and `sdb`:

    [Thu Jan  1 02:00:07 1970] sd 1:0:0:0: [sdb] 312581808 512-byte logical blocks: (160 GB/149 GiB)
    [Thu Jan  1 02:00:07 1970] sd 1:0:0:0: [sdb] Write Protect is off
    [Thu Jan  1 02:00:07 1970] sd 1:0:0:0: [sdb] Mode Sense: 34 00 00 00
    [Thu Jan  1 02:00:07 1970] sd 1:0:0:0: [sdb] Write cache: disabled, read cache: enabled, doesn't support DPO or FUA
    [Thu Jan  1 02:00:07 1970]  sdb: sdb1
    [Thu Jan  1 02:00:07 1970] sd 1:0:0:0: [sdb] Attached SCSI disk
    [Thu Jan  1 02:00:08 1970] fuse init (API version 7.26)
    [Thu Jan  1 02:00:09 1970] random: crng init done
    [Thu Jan  1 02:00:10 1970] sd 0:0:0:0: [sda] 1250263728 512-byte logical blocks: (640 GB/596 GiB)
    [Thu Jan  1 02:00:10 1970] sd 0:0:0:0: [sda] Write Protect is off
    [Thu Jan  1 02:00:10 1970] sd 0:0:0:0: [sda] Mode Sense: 47 00 00 08
    [Thu Jan  1 02:00:10 1970] sd 0:0:0:0: [sda] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA
    [Thu Jan  1 02:00:11 1970]  sda: sda1
    [Thu Jan  1 02:00:11 1970] sd 0:0:0:0: [sda] Attached SCSI disk

One of them (`sda`) contains a partition with a valid `ext4` filesystem:

    [Thu Jan  1 02:00:52 1970] EXT4-fs (sda1): couldn't mount as ext3 due to feature incompatibilities
[Thu Jan  1 02:00:52 1970] EXT4-fs (sda1): couldn't mount as ext2 due to feature incompatibilities
[Thu Jan  1 02:00:55 1970] EXT4-fs (sda1): mounted filesystem with ordered data mode. Opts: (null)

What does your storage drop down menu look like after booting with this configuration?

jpihtila, 2018-03-11:
Now there is only one USB storage (sda). I did take other one away.
Sorry, i'm noob. This is first time with Raspberry. And no experience with Linux.
How I change configuration?

chrisf4lc0n, 2018-03-11:
There are definitely 2 drives there:
1. 160GB
2. 640GB
Unless you have got some enormous 160GB memory card you have installed the OS on.

jpihtila, 2018-03-11:
Yes there was 2 drives.
i did remove first one (160GB)

chrisf4lc0n, 2018-03-11:
What does the drop down "Storage Device" menu say?

jpihtila, 2018-03-11:
USDU1
Custom Path
Network share

chrisf4lc0n, 2018-03-11:
Can you ssh to your Raspi 3 and issue command issue command mount -l and fdisk -l?

jpihtila, 2018-03-11:
[image]

chrisf4lc0n, 2018-03-11:
mount -l
fdisk -l
please

jpihtila, 2018-03-11:
then what?
Sorry. I don't have clue.

chrisf4lc0n, 2018-03-11:
Then post the results.

jpihtila, 2018-03-11:
[image]

chrisf4lc0n, 2018-03-11:
"l" for Lima.

jpihtila, 2018-03-11:
now i have different disk in Raspberry
[image]

jpihtila, 2018-03-11:
sor
[image]
ry wrong picture

jpihtila, 2018-03-11:
i did format that disk ext4

chrisf4lc0n, 2018-03-11:
OK so the drive is there and hopefully mounted.
Isn't USDU1 your external drive?

jpihtila, 2018-03-11:
No it's memory card. disk usage 0/14,5GB
It's 16GB card

chrisf4lc0n, 2018-03-11:
Can I see the output of the mount -l? If the command in MotionEyeOS exists.

jpihtila, 2018-03-11:
nothing happening with command mount -l

chrisf4lc0n, 2018-03-11:
The sda1 file system does not look right... if it was EXT4 it would come up as "Linux" or Extended" not HPFS/...

chrisf4lc0n, 2018-03-11:
You are 100% positive that the whole drive has been formatted as EXT4?

jpihtila, 2018-03-11:
i did format that disk with mini tool partition wizard, and i did change system to EXT4.
But i can try again.

chrisf4lc0n, 2018-03-11:
[https://kwilson.io/blog/format-a-linux-disk-as-ext4-from-the-command-line/](https://kwilson.io/blog/format-a-linux-disk-as-ext4-from-the-command-line/)

jpihtila, 2018-03-11:
That format was perhaps too fast.  I try that. 
Thank you.

jpihtila, 2018-03-11:
Now type is Linux
[image]

chrisf4lc0n, 2018-03-11:
Is the drive in drop down "Storage Device" menu now?

jpihtila, 2018-03-11:
No

jpihtila, 2018-03-11:
Any idea what try next?
I did everything again. 1. Format memorycard 2. Install motion eye os
Everything else working fine. Motion detection, Gmail notification and google drive upload.
But not the hard drive. Camera is in Raspberry pi zero w. I try make this raspberry pi3 to function as hub.

jpihtila, 2018-03-13:
Drive is in drop down "storage device"  menu, when camera is connected in raspberry pi3. Mass storage option disappears when i move camera to pi zero w and reboot pi3.
Is there possibility to add mass storage without camera in pi3?

JamBalaya56562, 2026-11-08:
Closing with an answer to the last question. The disk showed up while the camera was attached to the Pi 3, and disappeared once that camera moved to the Pi Zero W and was added to the Pi 3 over the network. The file storage settings belong to the motionEye that has the camera, here the Pi Zero, so they offer its disks, not those of the Pi 3. To store on the Pi 3's disk, the Pi Zero can write to a network share on it.

closed 2026-11-08 by JamBalaya56562