motioneye-project/motioneye #823: Camera not usable - WARNING: Connect error on fd 15: ECONNREFUSED

cuddylier, 2018-03-12:
I always had no problem with motioneyeos and my Raspberry pi camera but one day my camera became physically disconnected, when I reconnected it and rebooted motioneyeos, the camera is no longer usable. It shows up as a connected device in '/dev/' under 'video0' but motion can't use it. Motioneyeos just shows this: https://image.ibb.co/gOeBTS/7e2Jby0i.png

The errors I get as shown in the motioneye.log are:

2018-03-11 19:15:35: [motioneye] ERROR: motion not running, starting it
2018-03-11 19:15:37: [motioneye] WARNING: Connect error on fd 15: ECONNREFUSED
2018-03-11 19:15:37: [motioneye] WARNING: Connect error on fd 15: ECONNREFUSED
2018-03-11 19:15:37: [motioneye] WARNING: Connect error on fd 15: ECONNREFUSED
2018-03-11 19:15:37: [motioneye] WARNING: Connect error on fd 15: ECONNREFUSED
2018-03-11 19:15:38: [motioneye] WARNING: Connect error on fd 15: ECONNREFUSED
2018-03-11 19:15:38: [motioneye] WARNING: Connect error on fd 15: ECONNREFUSED
2018-03-11 19:15:38: [motioneye] WARNING: Connect error on fd 15: ECONNREFUSED
2018-03-11 19:15:38: [motioneye] WARNING: Connect error on fd 15: ECONNREFUSED
2018-03-11 19:15:38: [motioneye] WARNING: Connect error on fd 15: ECONNREFUSED
2018-03-11 19:15:39: [motioneye] WARNING: Connect error on fd 15: ECONNREFUSED
2018-03-11 19:15:39: [motioneye] WARNING: Connect error on fd 15: ECONNREFUSED
2018-03-11 19:15:39: [motioneye] WARNING: Connect error on fd 15: ECONNREFUSED
2018-03-11 19:15:39: [motioneye] WARNING: Connect error on fd 15: ECONNREFUSED
2018-03-11 19:15:39: [motioneye] WARNING: Connect error on fd 15: ECONNREFUSED
2018-03-11 19:15:40: [motioneye] WARNING: Connect error on fd 15: ECONNREFUSED
2018-03-11 19:15:40: [motioneye] WARNING: Connect error on fd 15: ECONNREFUSED
2018-03-11 19:15:41: [motioneye] WARNING: Connect error on fd 15: ECONNREFUSED
2018-03-11 19:15:41: [motioneye] WARNING: Connect error on fd 15: ECONNREFUSED
2018-03-11 19:15:41: [motioneye] WARNING: Connect error on fd 15: ECONNREFUSED
2018-03-11 19:15:41: [motioneye] WARNING: Connect error on fd 15: ECONNREFUSED
2018-03-11 19:15:41: [motioneye] WARNING: Connect error on fd 15: ECONNREFUSED

It seems to keep saying that motion isn't running every so often.

The motion.log nor any of the other logs have anything notable in them.

I'm running version 20180224.

I tested a full reinstall and it made no difference.

I posted here rather than on motion because I was told to do so on this issue: https://github.com/Motion-Project/motion/issues/672

Thanks

Mrnt, 2018-03-15:
Have you checked that the camera is working outside of MotionEye - for example tested it with raspistill?

Zortris, 2018-04-06:
Same here. Camera(s) working correctly outside MotionEye. After restart the raspberry it work for some minutes or hours! I just want to use it for 24/7 recording on a NAS. Worked perfect till a few weeks ago.

Zortris, 2018-04-27:
Push

roger-zacharczyk, 2018-06-28:
Having the same problem. Occurs when camera detects motion and tries to write movie file. Of course the directory is empty - no movie files there.

EugeneSkrebnev, 2020-04-25:
have same problem with nanopi duo2

buteomont, 2022-02-23:
Same problem.  Anyone figure it out?

JamBalaya56562, 2026-11-10:
Closing, as this got no follow-up and was with a 2018 motionEyeOS image. `ECONNREFUSED` there only means motionEye could not reach Motion, which had failed to start with the reconnected camera; `motion.log` shows why. Please open a new issue with it if this still happens with a current release.

closed 2026-11-10 by JamBalaya56562