motioneye-project/motioneye #826: reduce light "spot"

martyouel, 2018-03-15:
Hi,

I added a new cam to my motioneye setup.
The cam is there to monitor my driveway. 

It work nice the day, but at night, because I have a floor lamp right in the direction where my cam is directed, I get only a "big white flash" on the picture.

Is there any setting I can use / try to reduce this ?

thanks in advance for your tips

[image]

jasaw, 2018-03-16:
Sounds like you need a hood over your camera lens.

JamBalaya56562, 2026-11-09:
As answered, this is about the light reaching the lens, so a hood or moving the camera helps more than any setting. Closing, as the question is answered.

closed 2026-11-09 by JamBalaya56562