motioneye-project/motioneye #827: Fatal Python error: GC object already tracked

pi0user, 2018-03-15:
Yesterday I installed motionEye on my x86 Debian 9 server following the Wiki Debian install guide. Installation of motion, ffmpeg & v4l-utils were skipped as I'm using motionEye as a hub for a few Pi Zero W's running the latest motionEyeOS. So, I basically started installation with step 3. Install the dependencies from the repositories.

The motionEye GUI can be accessed directly by http://ip:8765 and by https through an apache reverse proxy. Configuration and adding the remote cameras went well but the motionEye GUI keeps quitting unexpectedly. Camera output stops and the "no camera" images are displayed. The cameras are still accessible through their own GUI's and working fine.

When I restart meyectl with the -d option, I see the following error after a while.

>  Fatal Python error: GC object already tracked
> Aborted

The full log is as follows.

> 2018-03-15 08:49:01: [motioneye]     INFO: hello! this is motionEye server 0.38.1
> 2018-03-15 08:49:01: [motioneye]     INFO: motion not installed
> 2018-03-15 08:49:01: [motioneye]     INFO: v4l-utils not installed
> 2018-03-15 08:49:01: [motioneye]    DEBUG: reading main config from file /etc/motioneye/motion.conf...
> 2018-03-15 08:49:01: [motioneye]    DEBUG: loading additional config structure for main, without separators
> 2018-03-15 08:49:01: [motioneye]    DEBUG: additional config section: network
> 2018-03-15 08:49:01: [motioneye]    DEBUG: listing config dir /etc/motioneye...
> 2018-03-15 08:49:01: [motioneye]    DEBUG: found camera with id 2
> 2018-03-15 08:49:01: [motioneye]    DEBUG: found camera with id 1
> 2018-03-15 08:49:01: [motioneye]    DEBUG: reading camera config from /etc/motioneye/thread-1.conf...
> 2018-03-15 08:49:01: [motioneye]    DEBUG: reading camera config from /etc/motioneye/thread-2.conf...
> 2018-03-15 08:49:01: [motioneye]     INFO: cleanup started
> 2018-03-15 08:49:01: [motioneye]     INFO: wsswitch started
> 2018-03-15 08:49:01: [motioneye]     INFO: tasks started
> 2018-03-15 08:49:01: [motioneye]     INFO: mjpg client garbage collector started
> 2018-03-15 08:49:01: [motioneye]     INFO: server started
> 2018-03-15 08:50:01: [motioneye]    DEBUG: running cleanup process...
> 2018-03-15 08:50:01: [motioneye]    DEBUG: cleaning up pictures...
> 2018-03-15 08:50:01: [motioneye]    DEBUG: cleaning up movies...
> 2018-03-15 08:50:01: [motioneye]    DEBUG: cleanup done
> 2018-03-15 09:06:05: [motioneye]    DEBUG: executing action "snapshot" of remote camera 1 on http://[removed]/config/1
> 2018-03-15 09:06:05: [motioneye]    DEBUG: 200 POST /action/1/snapshot/?_=1521101166057&_username=admin&_signature=[removed] (127.0.0.1) 304.89ms

This time motionEye aborted after taking a snapshot. It also seems to abort after a period of inactivity with no apparent reason or trigger.

Any ideas on how to debug & resolve this issue?

motionEye server 0.38.1
Python 2.7.13
Linux Server 4.9.0-6-686-pae #1 SMP Debian 4.9.82-1+deb9u3 (2018-03-02) i686 GNU/Linux

ccrisan, 2018-03-15:
Sounds like a python problem rather than a motionEye problem.

pi0user, 2018-03-15:
Thanx. I'll look into my servers python configuration.

pi0user, 2018-03-16:
The system is running smoothly for 24 hours after an upgrade of motionEye.
`pip install motioneye --upgrade`

That (re)installed some stuff and I guess fixed whatever was wrong.
I put that command in a cron job to keep up to date with motionEye and its dependencies.

@ccrisan, many thanks for this great, feature packed software!

closed 2018-03-16 by pi0user