motioneye-project/motioneye #828: Motioneye on stretch no run-command notification

Steinkauz80, 2018-03-16:
Hello,
my goal is to trigger a DSLR camera with flashlight on detected motion with raspberry and IR-camera at nighttime. 
I have installed 2017-11-29-raspbian-stretch.img on a raspberry pi3.
Then I installed motion and motioneye following the instructions on
https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian
Finally I took a little Python script to switch a GPIO-pin.
Motion is detected. Pictures are stored in the Raspi and also the Email-notification works fine.

But the „run a command“ does not work at all. Even when I write it including  the command and full path in the Motioneye command field:     python /home/pi/trigger18.py

Running trigger18.py from raspi-cli or via ssh works fine.
I hope anyone can give some hints.
Many thanks in advance.

fuzzyskibum, 2018-03-21:
Having the same issue. Running up to date MotionEye with same Raspbian image. I'm attempting to play an audio file from audio output jack on RPi3 using simple shell script saved in /home/pi/. Script runs fine from terminal and audio plays. Any help greatly appreciated! Thanks.

fuzzyskibum, 2018-03-28:
Steinkauz80 I fixed this issue in my project by working through the resolution found in this previous post:  https://github.com/ccrisan/motioneye/issues/74.  As in the discussion, I realized that motioneye was starting from /etc/rc.local and thus runs with root and not pi. Shell script specified in Run Command (stored as /root/file.sh) is now functioning properly for me. Hope this helps!

closed 2022-03-19 by MichaIng