motioneye-project/motioneye #833: Google drive upload slows down

kantorbalazs, 2018-03-17:
Hello!

I had 1 camera with motioneye on a rapsberry pi, with the fuction of uploading pictures taken every 10 seconds. I have installed a second camera, applied the same setup, and now, the google drive upload started to slow down. At this point i have 40 minutes of delay(and increasing) between real life and google drive upload. Can you help, what should I check or modify in order to make this work?
With one camera everything was OK, and appeared on almost real time.

cybercorey, 2018-04-16:
After brief look into this, it looks like all the uploads get added to a queue and then uploaded one by one. The more images to upload the longer this takes. Meaning that the queue starts piling up and this is causing the out of sync issues. 
I'm no python Dev, but I think allowing mutilthreading in the queue would help. Meaning that it can process more than one upload at a time.

If I get some time I'll look into testing this idea.

darkplatinum, 2018-06-24:
I'm having issues with google drive uploading. Google drive kills bandwidth. I think google added a bandwidth limiter some time ago but not sure if it's possible for us to use. Can't seem to find an option in google drive.