motioneye-project/motioneye #834: Unable to open Video Device, when it's dark outside

Benni83, 2018-03-17:
Hi,
I'm a beginner with Motioneye. Its a really nice OS. I'm using it to transform signal from Doorbell to use it my smart home, which can only handle mjpeg. 
And although for security recordings.
At daylight it works really nice! But when it gets dark outside, i get a problem.
Then I have only a grey image, which says "unable to open Video Device".
When light comes back in the morning, everything works fine again.
Does anyone have an idea, what i could check or set, to prevent this problem?
If I open camera stream directly with vlc or other program, ist dark, but you can recognise a few things. so camera works probably.

Thanks and best regards

Benni83, 2018-03-17:
[image]
Just some more information. At night, image looks like this.

Camera is connected via ip.

There is no light source, or ir light. Only a street lamp a view meters away

LiaMaps, 2018-04-06:
Got the same issue but think it is more a timeout problem than a "dark" problem. My motioneye runs on a seperate ubuntu machine. I got 4 raspberry camera as network cameras connected (i installed rpi cam control on all) and get the cam pics (mjpeg i think) with http://xxx.xxx.xxx.xxx/html/cam.jpg everything runs smooth but at some stage the motioneye times out the cam and after this not connecting it again. Service restart oder save the config via admin menu brings the cam back up again.

LiaMaps, 2018-04-09:
This error is a million times in the log if i put the log level to "error":

> [4] [ERR] [NET] netcam_read_first_header: Error sending 'connect' request: Broken pipe

So it seems my motion is too old. 

> »motion« ist bereits die neuste Version (3.2.12+git20140228-8build1).

iam on ubuntu LTS and will try to force update motion.

LiaMaps, 2018-04-09:
too early too happy. same problem. will dicuss this with motion developers.

Benni83, 2018-04-11:
By me still same. Connection lost, when it’s getting dark. In the morning video comes back. Because of spring, video stays longer every day and comes back earlier in the morning. Now since a view week. Changing settings don’t helps.

tech-nickel, 2021-07-28:
Is there any update? I'm running the latest
motioneye 0.42.1 with motion 4.3.2 on Raspbian GNU/Linux 10 (buster).
Still the same problem when trying to get a JPEG Stream from Rpi Cam Web Interface with authentication. After a few seconds the stream freezes and a few minutes later I get "Connection to camera lost" with a bulk of error messages from motioneye `[1:nc2:Camera1] [ERR] [NET] netcam_read_first_header: Error sending 'connect' request: Broken pipe`.
Im glad for any help.

starbasessd, 2021-07-28:
I don't know anything about RPi Cam Web Interface. MotionEyeOS is an extremely lightweight dedicated motionEye on just enough OS to work. There is a Fast Network Camera option if you are using the PiCam (CSI Ribbon Cable) to lessen CPU usage even further. There are versions for 14 different boards available. 
motionEye on another OS like Raspbian or Ubuntu or whatever and motionEyeOS work very well together. That would be my first stop. motionEyeOS supports any camera that RaspberryPiOS supports. motionEye on RaspberryPiOS is very comprehensive, too.

starbasessd, 2021-07-28:
BTW, this issue is over 3 years old. It would be better to start a new one.

tech-nickel, 2021-08-19:
Sorry for the late reply. Thank you for your answer, I'will try the option with motionEye on Raspberry Pi OS.

JamBalaya56562, 2026-11-08:
Closing, as this is from 2018 and 2021 with motionEye 0.42.1 or older. In the dark, many cameras lower their frame rate or send frames late for longer exposure, and Motion gives up on a network camera that sends nothing within its timeout, which shows as "unable to open video device" until frames come again. A current Motion, or a camera set to keep a fixed frame rate at night, is the way out; please open a new issue with `motion.log` if it still happens with a current release.

closed 2026-11-08 by JamBalaya56562