motioneye-project/motioneye #838: Different resolutions for video and still image

hoover123, 2018-03-21:
I have set the video resolution to 1920x1080. Videos are captured in 1920x1080 but still images only arrive at 800x600. I would like to have the still images also in 1920x1080.

`v4l2-ctl --list-formats
ioctl: VIDIOC_ENUM_FMT
	Index       : 0
	Type        : Video Capture
	Pixel Format: 'MJPG' (compressed)
	Name        : Motion-JPEG

	Index       : 1
	Type        : Video Capture
	Pixel Format: 'YUYV'
	Name        : YUYV 4:2:2

v4l2-ctl --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
	Index       : 0
	Type        : Video Capture
	Pixel Format: 'MJPG' (compressed)
	Name        : Motion-JPEG
		Size: Discrete 1920x1080
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.040s (25.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
		Size: Discrete 1280x720
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.040s (25.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
		Size: Discrete 800x600
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.040s (25.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
		Size: Discrete 640x480
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.040s (25.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
		Size: Discrete 640x360
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.040s (25.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
		Size: Discrete 352x288
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.040s (25.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
		Size: Discrete 320x240
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.040s (25.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
		Size: Discrete 1920x1080
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.040s (25.000 fps)
			Interval: Discrete 0.067s (15.000 fps)

	Index       : 1
	Type        : Video Capture
	Pixel Format: 'YUYV'
	Name        : YUYV 4:2:2
		Size: Discrete 640x480
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.040s (25.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
		Size: Discrete 800x600
			Interval: Discrete 0.067s (15.000 fps)
		Size: Discrete 640x360
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.040s (25.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
		Size: Discrete 352x288
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.040s (25.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
		Size: Discrete 320x240
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.040s (25.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
		Size: Discrete 640x480
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.040s (25.000 fps)
			Interval: Discrete 0.067s (15.000 fps)`

ccrisan, 2018-03-21:
This is the first time I've heard of such an issue. Please prepare the configurations and logs necessary to [report a motion issue](https://github.com/ccrisan/motioneye/wiki/Report-Motion-Issue) and then ask them if there's anything odd in your configuration that might make motion generate still images of different resolution.

hoover123, 2018-03-23:
I just verified with
fswebcam -r 1920x1080 image.jpg
that the USB camera module is capable of producing still images at 1920x1080. Yes it does.

Please note in above listing the maximum resolutions of the usb camera firmware:

Pixel Format: 'MJPG' (compressed)
Name : Motion-JPEG
Size: Discrete 1920x1080

Pixel Format: 'YUYV'
Name        : YUYV 4:2:2
Size: Discrete 800x600

Somehow it seems to default to YUYV when taking still images thus 800x600. However, it should use MJPG with 1920x1080 maximum.

ccrisan, 2018-03-25:
@hoover123 the (v4l2_palette)[https://rawgit.com/Motion-Project/motion/master/motion_config.html#v4l2_palette] Extra Motion Option might help you.

hoover123, 2018-03-25:
v4l2_palette=8
in Extra Motion Options produces still images at 1920x1080 now. This setting is the equivalent of MJPG.
However, the capture frame rate drops down to 0.2 - 0.5 fps on average, which is probably due to the high bandwidth requirements of MJPG.
But thanks for that hint.

ccrisan, 2018-03-25:
@hoover123 at this point I honestly don't think there's anything else you or I can do to improve the performance. Can we consider the issue resolved?

hoover123, 2018-03-26:
Yeah, that‘s fine. Thanks again.

closed 2018-03-26 by ccrisan