motioneye-project/motioneye #840: Extremely low FPS on Raspbian armhfp & Devuan ARM64.

kenny-w7, 2018-03-25:
So I've been testing things out on my Raspberry Pi 3B & it would appear it makes no difference what Linux distro I use, the FPS is always under 5 fps... Can't make it go any higher, I've even tried Fedora armhfp, same results, stuck at 5 FPS. I've changed the settings in the config files & on the web interface but it still won't go above 5 FPS, and when theres actual motion going on, it goes even lower! I've tried using different versions of motion & motioneye, nothing seems to make it work properly....

ccrisan, 2018-03-25:
Have you tried reducing the resolution? Or maybe changing the camera you use?

kenny-w7, 2018-03-25:
I've got it at 352x288, doesn't matter whether its on that or if its on 854x480 or anywhere in-between, and its currently the only camera I have, I can get it to go past 5 fps on guvcview... I've tried changing the FPS rates & it doesn't seem to make any difference whether its on 10, 15, 20, 25, 45, etc..

ccrisan, 2018-03-25:
Are you sure you have set both the capture framerate and streaming framerate to something above 5fps?

kenny-w7, 2018-03-26:
Yep, both are at 25 currently

ccrisan, 2018-03-26:
@kenny-w7 how about your preferences? Do you have a framerate dimmer?

kenny-w7, 2018-03-31:
A framerate dimmer? I don't think so...

ccrisan, 2018-04-01:
@kenny-w7 the Framerate Dimmer is a preference in the motionEye panel (one of the very first settings).

tasteeeeee, 2018-04-11:
@drphil3d hi! can you talk how use omx?i bought mini-pc with intell n3510 and 8gb ram and have 0-3 fps on camera. I don`t think that pc have low fps, thanks!
https://www.aliexpress.com/item/XCY-Barebone-Mini-PC-Intel-Celeron-N2930-N2940-N3510-Quad-cores-Micro-Desktop-Computer-Household-Nettops/32817027731.html?spm=a2g0v.10010108.1000016/B.1.a0a82712EWa0d2&isOrigTitle=true

tasteeeeee, 2018-04-11:
in raspberry pi i have 16+ fps :)

JamBalaya56562, 2026-11-09:
With the hints here the frame rate went up to 8-12 fps. A USB camera's frame rate often also depends on its own exposure in low light and on whether it delivers MJPEG. Closing, as there was no news since 2018; if it is still low on a current version, please open a new issue with the camera model and the motion log.