motioneye-project/motioneye #843: not save image

regolo76, 2018-03-26:
Hi, I installed motioneye but it does not save images inside the diretory 

/var/lib/motioneye (via browser: Root Directory /var/lib/motioneye/CameraX)
The directory has 777 permissions.

log motion:
[2: ml2] [WRN] [NET] Above message repeats 1 times
[2: ml2] [WRN] [NET] netcam_init_jpeg: no new pic, no signal rcvd
[4: ml4] [WRN] [NET] netcam_init_jpeg: no new pic, no signal rcvd

log motioneye:
not exists in /var/log/

motioneye.conf
# path to the directory where log files go (must be writable by motionEye)
log_path /var/log

Thank you

mycomm, 2018-04-08:
You need to configure 
  Still Images
      and
  File Storage

If you don't set up 'Still Images', motioneye will trigger, but will not write the images to file.

SarenT, 2020-07-17:
I have a similar issue. Some images are saved but not all. Most importantly, manual snapshots are also not saved.

SarenT, 2020-07-17:
After more investigation, it seems that motion or motioneye is ignoring the File Storage/Root Directory setting, which was:
/var/lib/motioneye/CameraX
All the triggers and snapshots were saved under /var/lib/motioneye/ instead.
After setting File Storage/Root Directory to  /var/lib/motioneye, all images (even previously not visible ones) were visible. I think the confusion was that this setting doesn't where to save but to find saved images for the pictures browser. What I don't get is, why some images were saved under /var/lib/motioneye/CameraX? I can't reproduce this at the moment.

zagrim, 2020-07-18:
@SarenT MotionEye looks for files (in media browser) under the Root Directory, so if that points below the level where some files are stored, those files won't be shown. Could it be by chance that you've changed Root Directory at some point in time, to not include `CameraX` anymore, then reverting that back later on? Or have you used `../` in Image File Name? That could produce similar results, too (I've not tested that that actually would be accepted be the code, though).

SarenT, 2020-07-18:
@zagrim It is possible. But I didn't use ../ for sure in any location. If then the CameraX. However, I have just created the new camera in motioneye. So even if I have changed in the past (can't remember really), this must have happened with a previous camera. I have recenetly upgraded motioneye to the newest version.

JamBalaya56562, 2026-11-10:
Closing, as this was answered: pictures are written only with Still Images enabled, and the media browser looks for them under the camera's Root Directory, so files saved elsewhere do not show there, as zagrim explained.

closed 2026-11-10 by JamBalaya56562