motioneye-project/motioneye #848: It doesnt work with the last update upgrade.

ConsGab, 2018-03-29:
$ sudo  apt-get install ffmpeg v4l-utils
Lecture des listes de paquets... Fait
Construction de l'arbre des dépendances
Lecture des informations d'état... Fait
v4l-utils is already the newest version (1.12.3-1).
Certains paquets ne peuvent être installés. Ceci peut signifier
que vous avez demandé l'impossible, ou bien, si vous utilisez
la distribution unstable, que certains paquets n'ont pas encore
été créés ou ne sont pas sortis d'Incoming.
L'information suivante devrait vous aider à résoudre la situation :

Les paquets suivants contiennent des dépendances non satisfaites :
 ffmpeg : Dépend: libavcodec57 (>= 7:3.2.10) mais ne sera pas installé ou
                   libavcodec-extra57 (>= 7:3.2.10) mais ne sera pas installé
          Dépend: libavdevice57 (>= 7:3.2.10) mais ne sera pas installé
          Dépend: libavfilter6 (>= 7:3.2.10) mais ne sera pas installé ou
                   libavfilter-extra6 (>= 7:3.2.10) mais ne sera pas installé
          Dépend: libavformat57 (>= 7:3.2.10) mais ne sera pas installé
          Dépend: libavresample3 (>= 7:3.2.10) mais ne sera pas installé
          Dépend: libavutil55 (>= 7:3.2.10) mais ne sera pas installé
          Dépend: libpostproc54 (>= 7:3.2.10) mais ne sera pas installé
          Dépend: libsdl2-2.0-0 (>= 2.0.4) mais il n'est pas installable
          Dépend: libswresample2 (>= 7:3.2.10) mais ne sera pas installé
          Dépend: libswscale4 (>= 7:3.2.10) mais ne sera pas installé
          Dépend: libva1 (>= 1.7.3) mais il n'est pas installable
E: Impossible de corriger les problèmes, des paquets défectueux sont en mode « garder en l'état ».

jsavage, 2018-03-29:
mmm. acually, i might have encountered this too. when attempting to install earlier today at the firat part when ffmpeg is installed. my error was in English!  so i cant compare exactly. i resolved it by performing an additional apt update and another dist-upgrade,  i wonder if this had anything to do with my issue #258 relating to motion.conf ?

ConsGab, 2018-03-30:
It's done.
I had to do again ALL the procédure provided by : 
[](https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian.)
Thank's

closed 2018-03-30 by ConsGab