motioneye-project/motioneye #850: [Feature Request] Support for multiple views

fredrik-rambris, 2018-03-31:
I currently have 6 network cameras configured in MotionEye (MotionEyeOS running on Rasberry Pi 3). I would like to group them into Outdoors and Indoors groups and have different tabs showing each group.

tokenwizard, 2019-03-20:
I would like to +1 this request. Or, if someone has a better solution. 

Currently I have a Raspberry Pi setup as a kiosk to just display the motioneye page on a16:9 Monitor. Four cameras display nicely with the 2-column setup and the fit vertically options. In total, I have 6 cameras configured and setup an "auto-scroll feature" using xdotool to page down and then page up again every 10 seconds:

while true; do 
	xdotool key Page_Down; 
	sleep 10;
	xdotool key Page_Up;
	sleep 10;
done

BUT I plan on adding more cameras and it would be great if I could set up dual monitors and have one set of cameras on one monitor and another on the other, or just have the Kiosk browser flip back and forth between tabs for each set of monitors.

Thanks!

MobileWAN, 2019-04-25:
I'm using 4 cameras now and would like them to show up on one screen with two rows of two columns. Currently, I can only get the full top two cameras and only the top half of the bottom cameras. I can also scroll to get the full bottom images, and half of the top. Does the way to resolve this involve changing the display pixel settings for the individual cameras?

MAGICCC, 2020-01-31:
I second this FR.
I have multiple cameras spreaded in different physical rooms/areas. Would be nice to group all outdoor and all indoor cameras for example.
For now I would need to have multiple motioneye instances running at the same time