motioneye-project/motioneye #851: I can't install motioneye to Rasberry PI 3B+ ?

alpozdemir, 2018-04-01:
I can't install motioneye to Rasberry PI 3B+ Im using minimal image based on Debian Stretch. Its normal?

ccrisan, 2018-04-01:
Unless you give us more details about your problem, we can't help you. What error do you get? What went wrong?

blackdot, 2018-04-06:
I faced propably similar/same issue on my clean raspbian minimal image. I saw something like:
`The headers or library files could not be found for zlib, a required dependency when compiling Pillow from source.`

Installation of `zlib1g-dev` package helped.

timrowledge, 2018-04-07:
Probably related to #855

ccrisan, 2018-04-07:
Added `zlib-dev` to deps on the wiki page. I have just followed the instructions on a fresh install and everything worked perfectly.

closed 2018-04-07 by ccrisan