motioneye-project/motioneye #853: upgrade error?

nuiler, 2018-04-03:
I still make an update in the frontend version 0.38
is displayed.

sudo pip install motioneye --upgrade
Collecting motioneye
  Downloading motioneye-0.39.tar.gz (758kB)
    100% |████████████████████████████████| 768kB 164kB/s
Requirement already up-to-date: jinja2 in /usr/local/lib/python2.7/dist-packages (from motioneye)
Collecting pillow (from motioneye)
  Downloading Pillow-5.1.0.tar.gz (14.3MB)
    100% |████████████████████████████████| 14.3MB 10kB/s
Requirement already up-to-date: pycurl in /usr/local/lib/python2.7/dist-packages (from motioneye)
Collecting tornado>=3.1 (from motioneye)
  Downloading tornado-5.0.1.tar.gz (504kB)
    100% |████████████████████████████████| 512kB 262kB/s
Requirement already up-to-date: MarkupSafe>=0.23 in /usr/local/lib/python2.7/dist-packages (from jinja2->motioneye)
Requirement already up-to-date: backports_abc>=0.4 in /usr/local/lib/python2.7/dist-packages (from tornado>=3.1->motioneye)
Collecting futures (from tornado>=3.1->motioneye)
  Downloading futures-3.2.0-py2-none-any.whl
Requirement already up-to-date: singledispatch in /usr/local/lib/python2.7/dist-packages (from tornado>=3.1->motioneye)
Requirement already up-to-date: six in /usr/local/lib/python2.7/dist-packages (from singledispatch->tornado>=3.1->motioneye)
Building wheels for collected packages: motioneye, pillow, tornado
  Running setup.py bdist_wheel for motioneye ... done
  Stored in directory: /root/.cache/pip/wheels/ff/71/e5/0f2816473c6dff1066d0db7ef108081c06a904afb0d019084b
  Running setup.py bdist_wheel for pillow ... done
  Stored in directory: /root/.cache/pip/wheels/1f/7f/91/4ffaae9e9681c72189fa211788eb26cacdce944e11371f6876
  Running setup.py bdist_wheel for tornado ... done
  Stored in directory: /root/.cache/pip/wheels/fd/0d/50/bb8bebaf371beed13f2502466cdadc9ef2afb3302e66a56a36
Successfully built motioneye pillow tornado
Installing collected packages: pillow, futures, tornado, motioneye
  Found existing installation: Pillow 5.0.0
    Uninstalling Pillow-5.0.0:
      Successfully uninstalled Pillow-5.0.0
  Found existing installation: tornado 4.5.3
    Uninstalling tornado-4.5.3:
      Successfully uninstalled tornado-4.5.3
  Found existing installation: motioneye 0.38
    Uninstalling motioneye-0.38:
      Successfully uninstalled motioneye-0.38
Successfully installed futures-3.2.0 motioneye-0.39 pillow-5.1.0 tornado-5.0.1
pi@Webcam:~ $ sudo systemctl restart motioneye

motionEye Version | 0.38
-- | --
Motion Version | 4.1
OS Version | Raspbian 9.4

ccrisan, 2018-04-03:
You're not stopping/starting the correct motionEye installation. You might installed it in the past with a different python version. Uninstall it completely, make sure it's not running any more, and start clean.

nuiler, 2019-03-18:
How can I upgrade, currently I have this version. motionEye Version 0.39.3
Motion Version 4.1
OS version Raspbian 9.4

ccrisan, 2019-03-20:
See https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian, point no. 9.
Also make sure to read the release notes of [0.40rc5](https://github.com/ccrisan/motioneye/releases/tag/0.40rc5).

closed 2019-03-20 by ccrisan