motioneye-project/motioneye #855: Unable to install MotionEye on Raspbian 2018-03-13

timrowledge, 2018-04-03:
Testing install on the latest Raspbian ready for 3B+ usage; using a load script that works perfectly on the prior Raspbian  - just a gathered up shell script made from your Raspbian install page.

The sudo apt-get update has been done, as has an upgrade cycle.

Running my script (attached as loadmotioneye.sh in the zip file) fails because of a whole load of unmet dependencies as shown in errorlog-1. I tried a second 'apt-get update' in case the upgrade had changed anything and it seemed to make a slight difference as shown in errorlog-2.

[M-EyeRaspbian0313-errors.zip](https://github.com/ccrisan/motioneye/files/1873293/M-EyeRaspbian0313-errors.zip)

Eek!

ccrisan, 2018-04-07:
@timrowledge have you installed `libcurl4-openssl-dev`?

ektus, 2018-04-07:
Trying to install manually on a zero, I encountered similar problems. Notably, the very first step of installing ffmpeg failed. libavdevices57 and libsdl12-2.0-0 were reported as unmet dependencies.

 Installing libcurl4.openssl-dev first, the installation of ffmpeg was successful.

timrowledge, 2018-04-07:
No, I hadn't installed any libcurl; at least not deliberately.

timrowledge, 2018-04-07:
OK; fresh install of 2018-03-13-raspbian-stretch.zip with Etcher.
Added netatalk so I could see it on my network.
Did *not* do apt-get update/upgrade.

First attempted install failed because of errors such as

> After this operation, 11.3 MB of additional disk space will be used.
> Do you want to continue? [Y/n] y
> Err:1 http://mirrordirector.raspbian.org/raspbian stretch/main armhf libcurl4-openssl-dev armhf 7.52.1-5+deb9u4
>   404  Not Found [IP: 2a00:1098:0:80:1000:75:0:3 80]
> Get:2 http://muug.ca/mirror/raspbian/raspbian stretch/main armhf libjpeg62-turbo-dev armhf 1:1.5.1-2 [181 kB]
> Err:4 http://mirrordirector.raspbian.org/raspbian stretch/main armhf libssl-dev armhf 1.1.0f-3+deb9u1
>   404  Not Found [IP: 2a00:1098:0:80:1000:75:0:3 80]
> Get:3 http://muug.ca/mirror/raspbian/raspbian stretch/main armhf libjpeg-dev all 1:1.5.1-2 [56.1 kB]
> Err:5 http://mirrordirector.raspbian.org/raspbian stretch/main armhf libssl-doc all 1.1.0f-3+deb9u1
>   404  Not Found [IP: 2a00:1098:0:80:1000:75:0:3 80]
> Fetched 237 kB in 1s (145 kB/s)                                             
> E: Failed to fetch http://mirrordirector.raspbian.org/raspbian/pool/main/c/curl/libcurl4-openssl-dev_7.52.1-5+deb9u4_armhf.deb  404  Not Found [IP: 2a00:1098:0:80:1000:75:0:3 80]
... and so on.

Next I tried the apt-get update to see if the would correct the mirrordirector settings as it so often has for OS updates in the past.
This install suceeded, which is good. System runs and connects from broswer, which is very good.

---------------------
So the question is why the `sudo apt-get upgrade` causes the install to fail, a question I truly can't help with. Presumably something that gets upgraded to a newer version?

@ccrisan please have a go at an install on a
fresh 2018-03-13 install
+ sudo apt-get update 
+ sudo apt-get upgrade

Doing my best to understand the errorlog-1 I sent you it looks to me as if the upgrade breaks things right at the

>  v4l-utils is already the newest version (1.12.3-1).

stage, which would be strange and way outside my range of expertise.

ccrisan, 2018-04-08:
@timrowledge :

This set of errors is normal, given an outdated apt package database - `apt-get update` usually solves them.

I don't think `sudo apt-get upgrade` caused any problems in your install. On the contrary, it seems to (and should) have solved them.

The `v4l-utils` thing is just a notice saying that the package is already present. Nothing to worry about.

So do you still have issues with your installation or can we close this issue?

careingemt, 2018-04-08:
i get the following errors :+1: 

Motoineye errors
EditDelete postReport this postQuote
Sun Apr 08, 2018 12:23 pm

............. Round 4 with fresh OS .............

sudo apt-get update

Get:1 http://archive.raspberrypi.org/debian stretch InRelease [25.3 kB]

completes without errors

........................................................

sudo apt-get upgrade

Reading package lists... Done
Building dependency tree

no errors

...............................................................

Instructions: 
MUST BE RAN WITH ROOT 

STEP 1

Install ffmpeg and v4l-utils:

sudo apt-get install ffmpeg v4l-utils

I GET THE FOLLOWING ERROR 

pi@raspberrypi:~ $ sudo apt-get install ffmpeg v4l-utils
Reading package lists... Done
Building dependency tree
Reading state information... Done
v4l-utils is already the newest version (1.12.3-1).
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
ffmpeg : Depends: libavdevice57 (>= 7:3.2.10) but it is not going to be installed
Depends: libsdl2-2.0-0 (>= 2.0.4) but it is not installable
E: Unable to correct problems, you have held broken packages.

ANY help would be greatly appreciated

timrowledge, 2018-04-08:
careingemt seems t obe seeing the same effect that hit me; the upgrade part is causing some change that then causes problems.
As reported yesterday, a clean 2018-03-13 + setup + `sudo apt-get update` + MotioneEye install works.

I'll do a fresh run and include the `sudo apt-get upgrade` after the update and before installing MotionEye and report back...

careingemt, 2018-04-08:
type in terminal like this ?

pi@raspberrypi:~ $ sudo apt-get update + MotioneEye install

careingemt, 2018-04-08:
hoping for some good news

careingemt, 2018-04-08:
QUOTE "ccrisan"

@timrowledge have you installed libcurl4-openssl-dev?

ill try that 

pi@raspberrypi:~ $ sudo apt-get install libcurl4-openssl-dev
Reading package lists... Done
Building dependency tree
Reading state information... Done
Suggested packages:
  libcurl4-doc libcurl3-dbg libidn11-dev libkrb5-dev libldap2-dev librtmp-dev
  libssh2-1-dev libssl1.0-dev | libssl-dev
The following NEW packages will be installed:
  libcurl4-openssl-dev
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 327 kB of archives.
After this operation, 999 kB of additional disk space will be used.
Get:1 http://raspbian.mirror.constant.com/raspbian stretch/main armhf libcurl4-openssl-dev armhf 7.52.1-5+deb9u5 [327 kB]
Fetched 327 kB in 1s (241 kB/s)
Selecting previously unselected package libcurl4-openssl-dev:armhf.
(Reading database ... 124697 files and directories currently installed.)
Preparing to unpack .../libcurl4-openssl-dev_7.52.1-5+deb9u5_armhf.deb ...
Unpacking libcurl4-openssl-dev:armhf (7.52.1-5+deb9u5) ...
Processing triggers for man-db (2.7.6.1-2) ...
Setting up libcurl4-openssl-dev:armhf (7.52.1-5+deb9u5) ...
pi@raspberrypi:~ $

im not sure whats next?

timrowledge, 2018-04-08:
OK, fresh 2018-03-13 install.
run `sudo apt-get update`
run `sudo apt-get upgrade`
run `sudo apt-get -y install ffmpeg v4l-utils`
This resulted in 

> Reading package lists... Done
> Building dependency tree
> Reading state information... Done
> v4l-utils is already the newest version (1.12.3-1).
> Some packages could not be installed. This may [...]

Since it didn't claim it was fatal I tried the next step;
run `sudo apt-get -y install libmariadbclient18 libpq5`
Which resulted in 

> Reading package lists... Done
> Building dependency tree       
> Reading state information... Done
> Package libmariadbclient18 is not available, but is referred to by another package.
> This may mean that the package is missing, has been obsoleted, or
> is only available from another source
> 
> E: Package 'libmariadbclient18' has no installation candidate
> E: Unable to locate package libpq5

So evidently something loaded as part of the apt-get _upgrade_ step is breaking the setup for installing MotionEye. Since it is pretty much standard to do the two steps of update and then upgrade whenever installing a new OS release in order to pick up any late bug-fixes, this is a fairly serious problem!

Just in case, I did a second apt-get update, something I've never seen suggested before; this did actually seem to load some stuff. Trying the `sudo apt-get -y install libmariadbclient18 libpq5` again  behaved quite differently and found the packages - so evidently the update fixed something.

`sudo wget https://github.com/Motion-Project/motion/releases/download/release-4.1.1/pi_stretch_motion_4.1.1-1_armhf.deb` worked.

Unfortunately, 
`sudo dpkg -i pi_stretch_motion_4.1.1-1_armhf.deb`
did not work - 

> Selecting previously unselected package motion.
> (Reading database ... 124737 files and directories currently installed.)
> Preparing to unpack pi_stretch_motion_4.1.1-1_armhf.deb ...
> Unpacking motion (4.1.1-1) ...
> dpkg: dependency problems prevent configuration of motion:
>  motion depends on libavdevice57 (>= 7:3.2.9); however:
>   Package libavdevice57 is not installed.
> 
> dpkg: error processing package motion (--install):
>  dependency problems - leaving unconfigured
> Processing triggers for systemd (232-25+deb9u2) ...
> Processing triggers for man-db (2.7.6.1-2) ...
> Errors were encountered while processing:
>  motion
> 

So - problem caused by upgrade, partly solved by a second update but a dependency is now incorrect.

That's about all I can work out.

careingemt, 2018-04-08:
Having some issuses though

careingemt, 2018-04-08:
Web page won't load when I type pi IP and :8081

careingemt, 2018-04-08:
In terminal if I stop and restart daemon
Then I can re-open webpage

ccrisan, 2018-04-11:
@careingemt don't use port 8081. Use port 8765 if you're on plain motionEye, or port 80 if you're on motionEyeOS.

careingemt, 2018-04-11:
because of my above issue .... i started over with a fresh sd card 

tried to reinstall motion eye fails

careingemt, 2018-04-11:
Last login: Mon Apr  9 02:02:28 2018

pi@raspberrypi:~ $ sudo apt-get install libmariadbclient18 libpq5 libavcodec57  libavformat57 libavutil55 libswscale4

 

Hit enter on my keyboard

Reading package lists... Done

	Building dependency tree

	Reading state information... Done

	Package libmariadbclient18 is not available, but is referred to by another package.

	This may mean that the package is missing, has been obsoleted, or

	is only available from another source

Package libpq5 is not available, but is referred to by another package.

	This may mean that the package is missing, has been obsoleted, or

	is only available from another source

E: Package 'libmariadbclient18' has no installation candidate

	E: Package 'libpq5' has no installation candidate

	pi@raspberrypi:~ $

careingemt, 2018-04-11:
here is with a fresh stretch os 

     I've been trying to get the  "pimylifeup webcam in minutes"  

https://pimylifeup.com/raspberry-pi-webcam-server/

working on my pi3 for the last 5 day's.

 

Follwoing instructions for STRETCH OS

1. First, install the following packages. This command will work both on the full and lite version of Raspbian Stretch.

sudo apt-get install libmariadbclient18 libpq5 libavcodec57  libavformat57 libavutil55 libswscale4
 

Last login: Mon Apr  9 02:02:28 2018

pi@raspberrypi:~ $ sudo apt-get install libmariadbclient18 libpq5 libavcodec57  libavformat57 libavutil55 libswscale4

 

Hit enter on my keyboard

Reading package lists... Done
Building dependency tree
Reading state information... Done
Package libmariadbclient18 is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

Package libpq5 is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'libmariadbclient18' has no installation candidate
E: Package 'libpq5' has no installation candidate
pi@raspberrypi:~ $
 

 

Any help would be appreciated

Fran

bilum, 2018-08-17:
has anyone got a fix for this?  

E: Package 'libavcodec57' has no installation candidate
E: Package 'libavformat57' has no installation candidate
E: Package 'libavutil55' has no installation candidate
E: Package 'libswscale4' has no installation candidate

ektus, 2018-08-23:
Did you try installing libcurl4.openssl-dev as per my post from April 7th?

avvy65, 2018-12-06:
After this:  pip install motioneye, no errors upto the next part:
I get as far as this:                                                                                                                                   cp /usr/local/share/motioneye/extra/motioneye.conf.sample /etc/motioneye/motioneye.conf
cp: cannot stat '/usr/local/share/motioneye/extra/motioneye.conf.sample': No such file or directory

I checked for the folders and file, and there was no /motioneye/extra folders nor the motioneye.conf.sample file.

There are no motioneye.conf and motioneye.conf.sample files present on the pi zero.                      What went wrong.

closed 2022-03-19 by MichaIng