motioneye-project/motioneye #857: Strange colours / flickering with my connected network cams

LiaMaps, 2018-04-06:
Sometimes i got strange colours in my motioneye web view and also in recorded images. The images on the camera itself do not have any problem and i thin it has to be issues with the internal motioneye gfx drivers?
My cameras connected are all network cameras (raspberry pis) and everyone shows sometimes strange colours. In the last year i do not had any problems.
I updated my system some weeks ago and got the problem after that.

Thanks for help
[image]

LiaMaps, 2018-04-09:
too early too happy. same problem. will dicuss this with motion developers.

k1ngstire, 2022-12-06:
For anyone who has this issue.

I had it - permanently when running MotionEye on Ubuntu 22.04

switched to Debian 11 Bullseye and dont have the issue anymore.

geezer500, 2023-02-06:
Quick update....
Latest version of (built) motion (4.5.1+git20230125-ab0a463) and motioneye (dev branch) 0.43.0 fixes my flickering cameras (Tapo)

JamBalaya56562, 2026-11-07:
Closing, as newer versions fixed this for the people above: a current Motion with motionEye 0.43 for geezer500, and moving from Ubuntu 22.04 to Debian 11 for k1ngstire. Please open a new issue if the colours still go wrong with a current release.

closed 2026-11-07 by JamBalaya56562