motioneye-project/motioneye #863: Action Buttons on MotionEye under raspbian

minos2026, 2018-04-16:
Hi!
I have used action buttons on motioneyeOS and it was not hard but, now I use motioneye on raspbian because I need to access the camera from other location. That place where the camera is located, I have a 4g modem, and I can not do the port forwarding, so I am using ngrok tunnels  and dataplicity to interact whit it but, the action buttons setup is a little bit different and, I don't understand how to run a script.
I follow the instructions for motioneyeOS, changing the folder to /etc/motioneye and create the light_on_id but is not working. The switch is on the frame but it do not run the script.
Is there a chance to add an example here:
https://github.com/ccrisan/motioneye/wiki/Action-Buttons
I am just an enthusiastic, and might need a wiki for dummies

ccrisan, 2018-04-16:
1. Please confirm that your script is actually `/etc/motioneye/light_on_1`.
2. Please confirm that your script is executable.
3. Does it work when you run it manually, using the same user that runs motionEye (normally `root`)?
4. When you run it via motionEye, what happens? Do you get an error? If yes, please attach your `motioneye.log` file (normally obtained using `journalctl -u motioneye > motioneye.log`.

minos2026, 2018-04-16:
Thanks Mr. Ccrisan!
When its run manually its says permit denied:
[image]
[image]
[image]

I have this light_on.py, running fine from terminal. I thought might be some cables but they are working

Log
abr 16 09:29:32 raspberrypi motion[491]: [17761816:motion] [NTC] [ALL] motion_startup: Motion 4.1.1 Started
abr 16 09:29:32 raspberrypi motion[491]: [17761816:motion] [NTC] [ALL] motion_startup: Logging to syslog
abr 16 09:29:32 raspberrypi motion[491]: [17761816:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
abr 16 09:29:32 raspberrypi motion[491]: [17761816:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
abr 16 09:35:55 raspberrypi meyectl[215]:  WARNING: 403 GET /login/?_=1523892954630&_username=_&_signature=364930a4808067e843350b92e0860f2eb3042268 (192.
abr 16 09:36:06 raspberrypi meyectl[215]:    ERROR: [Errno 8] Exec format error
abr 16 09:36:06 raspberrypi meyectl[215]: Traceback (most recent call last):
abr 16 09:36:06 raspberrypi meyectl[215]:   File "/usr/local/lib/python2.7/dist-packages/tornado/web.py", line 1499, in _stack_context_handle_exception
abr 16 09:36:06 raspberrypi meyectl[215]:     raise_exc_info((type, value, traceback))
abr 16 09:36:06 raspberrypi meyectl[215]:   File "/usr/local/lib/python2.7/dist-packages/tornado/web.py", line 1700, in wrapper
abr 16 09:36:06 raspberrypi meyectl[215]:     result = method(self, *args, **kwargs)
abr 16 09:36:06 raspberrypi meyectl[215]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/handlers.py", line 1727, in post
abr 16 09:36:06 raspberrypi meyectl[215]:     self.run_command_bg(command)
abr 16 09:36:06 raspberrypi meyectl[215]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/handlers.py", line 1730, in run_command_bg
abr 16 09:36:06 raspberrypi meyectl[215]:     self.p = subprocess.Popen(command, stderr=subprocess.STDOUT, stdout=subprocess.PIPE)
abr 16 09:36:06 raspberrypi meyectl[215]:   File "/usr/lib/python2.7/subprocess.py", line 390, in __init__
abr 16 09:36:06 raspberrypi meyectl[215]:     errread, errwrite)
abr 16 09:36:06 raspberrypi meyectl[215]:   File "/usr/lib/python2.7/subprocess.py", line 1024, in _execute_child
abr 16 09:36:06 raspberrypi meyectl[215]:     raise child_exception
abr 16 09:36:06 raspberrypi meyectl[215]: OSError: [Errno 8] Exec format error
abr 16 09:36:06 raspberrypi meyectl[215]:    ERROR: 500 POST /action/1/light_on/?_=1523892965636&_username=admin&_signature=08487bd8a8e0962ee575fbb5c833b
abr 16 09:36:42 raspberrypi motion[491]: [1:ml1] [WRN] [ENC] ffmpeg_set_codec: Preferred codec h264_omx has been blacklisted

ccrisan, 2018-04-16:
You did not help me with a distinct answer for my 4 questions. However, this is clearly a problem of permissions. Use `sudo` do do changes to your script. Also you have a typo: you probably mean `value` instead of `valu`.

closed 2018-04-16 by ccrisan