motioneye-project/motioneye #864: docker mmal camera support

jf10, 2018-04-20:
Hello,

I am using the docker version of your repo (dev-armhf) on raspberry pi 3 with the stretch lite kernel. I use the following command to run the container : 

`docker run --name="motioneye" \
    -device=/dev/vchiq
    -p 8765:8765 \
    --hostname="motioneye" \
    -v /etc/localtime:/etc/localtime:ro \
    -v /etc/motioneye:/etc/motioneye \
    -v /var/lib/motioneye:/var/lib/motioneye \
    --restart="always" \
    ccrisan/motioneye:dev-amdhf`

But once in the browser it seems that motioneye does not detect the mmal camera. Did I miss something ?

kleini, 2018-04-20:
I never tried that. I have some Raspberry Pis but no mmal camera. Will have a look, where I can order a cheap one to test that scenario.

MrSimonC, 2018-07-09:
I also have this problem. Official Raspberry pi noir cameras can be bought from amazon or from [cpc farnell](http://cpc.farnell.com/c/education-maker-hobbyist-electronics/single-board-computers-microcontrollers/raspberry-pi/raspberry-pi-cameras-accessories/raspberry-pi-cameras) who do also [ship outside the uk](http://cpc.farnell.com/help-delivery-information#collapse-non-uk-address) if you need them to.

I'm experiencing the same issue as #871 making it very hard to get anything working on the raspberry pi. I've tried both the dev and master tagged docker images for armhf. Currently I'm using raspberry pi B+ with Noir V1 camera with latest raspbian (debian stretch). Without docker, I've gotten everything working fine with a native motioneye install.

Stopping the native motioneye service and trying docker, using someone else's non-constantly-restarting image using this command `sudo docker run --restart=always --name="motioneye" -p 8765:8765 --device=/dev/video0 -v /home/pi/motioneye:/var/lib/motioneye fafex/rpi-motioneye` I've found that i can login, but when trying to "add a camera", it has issues finding the /dev/video0 camera passed in - as `subprocess.Popen(['v4l2-ctl', '--list-devices']` must bring back nothing:

```python
ERROR: [Errno 2] No such file or directory
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/tornado/web.py", line 1467, in _stack_context_handle_exception
    raise_exc_info((type, value, traceback))
  File "/usr/local/lib/python2.7/dist-packages/tornado/web.py", line 1669, in wrapper
    result = method(self, *args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/motioneye/handlers.py", line 275, in get
    self.list()
  File "/usr/local/lib/python2.7/dist-packages/motioneye/handlers.py", line 205, in wrapper
    return func(self, *args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/motioneye/handlers.py", line 649, in list
    cameras = [{'id': d[1], 'name': d[2]} for d in v4l2ctl.list_devices()
  File "/usr/local/lib/python2.7/dist-packages/motioneye/v4l2ctl.py", line 53, in list_devices
    p = subprocess.Popen(['v4l2-ctl', '--list-devices'], stdout=subprocess.PIPE, bufsize=1)
  File "/usr/lib/python2.7/subprocess.py", line 710, in __init__
    errread, errwrite)
  File "/usr/lib/python2.7/subprocess.py", line 1335, in _execute_child
    raise child_exception
OSError: [Errno 2] No such file or directory
```
since this is a different image, obviously it may not be exactly the same issue, but it's too odd that both the ccrisan docker image and the one I've found are showing the same error - so it *might* help with this issue?

Anything else I can check, please let me know.

sujitrp, 2018-07-20:
This looks bug to me . With docker raspberry pi 3 MMAL camera is not working . any one found any solution  ?

marcelerkel, 2018-08-12:
@sujitrp If the problem you encounter is the fact that the camera is not found then I found the solution here: https://github.com/ccrisan/motioneye/issues/745

To apply this fix, enter the motioneye container:

    docker exec -ti motioneye /bin/bash

and execute the following command:

    ln -sf /opt/vc/bin/vcgencmd /usr/bin/vcgencmd

I'm forwarding the camera using the --device=/dev/vchiq option (not using --device=/dev/video0) and I am able to take still pictures. I haven't done any further testing. The host operating system I use is LibreELEC 8.2.5 on a Raspberry Pi 3.

@kleini Could you please add the above statement to the Dockerfile or add the /opt/vc/bin path to the $PATH variable so that any future images contain this fix? Thanks in advance.

kleini, 2018-08-20:
Will add that to the docker image.

fguiet, 2019-07-26:
Raspberry Pi Zero W + Raspberry Pi Cam + Docker + Dockerfile with "FROM resin/rpi-raspbian:stretch + ln -sf /opt/vc/bin/vcgencmd /usr/bin/vcgencmd" 
Everything is working great!

kleini, 2019-07-27:
Fixed with merge of #1384

closed 2019-07-27 by kleini