motioneye-project/motioneye #866: Cameras stop recording, but I still see them

minos2026, 2018-04-23:
Hi!
After some time learning (and testing) how to install and run  motioneye (with raspbian), the camera suddenly stopped recording. 

For background, the house where this camera is installed is alone from Monday to Friday and i am not there at the moment. Yesterday I left the house at 6pm (where the camera is installed), and based on what i was able to see while driving, the camera was recording fine. The camera was still recording today at 5 am, which is when i turned off a light (I love action buttons).

However, although i can still see the cameras and access them through dataplicity, take snapshots, etc., the camera is not recording any more. There is plenty of memory,  what can i do from my current location? As a side note, today in the morning I could not access the camera, I  guess the electric power was gone for an hour or so.; could that have affected the camera's ability to record?

minos2026, 2018-04-24:
I had have this issue in the, past  and reinstall everything again (is the way I fix things... :p) But now maybe with remote access (I use motioneye on raspbian), i can uninstall it and install it again... Can that fix it? Can be as easy as "pip uninstall motoneye"?

minos2026, 2018-04-24:
I did the "pip uninstall" and reinstall it but, the parameters remain, I guess I have to delete some other folders to completely delete everything because once was installed again, everything was there as before, my email, my little tweaks, everithing.
Then I change some things commented here: https://github.com/ccrisan/motioneye/issues/743
And is recording again

JamBalaya56562, 2026-11-07:
Closing, as you got it recording again with the changes from #743. Please open a new issue with `motion.log` if recording stops again with a current release.

closed 2026-11-07 by JamBalaya56562