motioneye-project/motioneye #867: Suddenly stoped recording

alan-shenton, 2018-04-24:
I have motionEye OS installed on a Raspberry Pi 3, attached to a 1TB USB HDD, with an IP camera.

I've set the recording to continuous 24/7, with files of 30 minutes in length, with automatic deletion for files older than one month. There is approx 30GB remaining on the drive, which is plenty for my needs.

This setup has been working for over 6 months now with no issues at all - which I am naturally thrilled about!

However... For some reason that I can't figure out, at approximately 5:21 pm GMT today the recordings failed. I didn't notice until I needed to check them for something! I now have one recording which is approx. a quarter of the size I would expect it to be - and which doesn't play, thus indicating a problem. This is along with 6 further recordings which are only 48 bytes each! Obviously, something has gone very wrong! (_Just when I bloody needed it as well!!!_)

I've rebooted the system, and it now appears to be working correctly. There are errors in the log file (attached) but I'm not well versed enough to know exactly what has gone wrong.

Here is an extract from the messages.log file (I can't attach it here as it's over 60mb!):

> Apr 24 17:12:46 CCTV-NVR user.notice throttlewatch: currently: arm-frequency-capped 
> Apr 24 17:13:16 CCTV-NVR user.notice throttlewatch: currently: arm-frequency-capped 
> Apr 24 17:13:46 CCTV-NVR user.notice throttlewatch: currently: arm-frequency-capped 
> Apr 24 17:14:16 CCTV-NVR user.notice throttlewatch: currently: arm-frequency-capped 
> Apr 24 17:14:46 CCTV-NVR user.notice throttlewatch: currently: arm-frequency-capped 
> Apr 24 17:15:16 CCTV-NVR user.notice throttlewatch: currently: arm-frequency-capped 
> Apr 24 17:15:46 CCTV-NVR user.notice throttlewatch: currently: arm-frequency-capped 
> Apr 24 17:15:47 CCTV-NVR user.debug motion: [1:ml1] [DBG] [EVT] exec_command: Executing external command '/usr/lib/python2.7/site-packages/motioneye/scripts/relayevent.sh "/data/etc/motioneye.conf" stop 1'
> Apr 24 17:15:47 CCTV-NVR user.debug motion: [1:ml1] [DBG] [EVT] exec_command: Executing external command '/usr/lib/python2.7/site-packages/motioneye/scripts/relayevent.sh "/data/etc/motioneye.conf" movie_end 1 /data/media/sda1/cctv/camera1/2018-04-24/Camera1 2018-04-24 1
> Apr 24 17:15:47 CCTV-NVR user.notice motion: [1:ml1] [NTC] [ALL] mlp_actions: End of event 1914
> Apr 24 17:15:47 CCTV-NVR user.debug motion: [1:ml1] [DBG] [EVT] exec_command: Executing external command '/usr/lib/python2.7/site-packages/motioneye/scripts/relayevent.sh "/data/etc/motioneye.conf" start 1'
> Apr 24 17:15:47 CCTV-NVR user.info motion: [1:ml1] [INF] [EVT] event_new_video: Source FPS 26
> Apr 24 17:15:47 CCTV-NVR user.info motion: [1:ml1] [INF] [ENC] ffmpeg_set_quality: libx264 codec vbr/crf/bit_rate: 35
> Apr 24 17:15:47 CCTV-NVR user.notice motion: [1:ml1] [NTC] [EVT] event_newfile: File of type 8 saved to: /data/media/sda1/cctv/camera1/2018-04-24/Camera1 2018-04-24 17-15-34.mp4
> Apr 24 17:15:47 CCTV-NVR user.notice motion: [1:ml1] [NTC] [ALL] motion_detected: Motion detected - starting event 1915
> Apr 24 17:15:58 CCTV-NVR user.info motion: [1:ml1] [INF] [ENC] ffmpeg_avcodec_log: specified frame type (3) at 0 is not compatible with keyframe interval
> Apr 24 17:16:16 CCTV-NVR user.notice throttlewatch: currently: arm-frequency-capped 
> Apr 24 17:16:46 CCTV-NVR user.notice throttlewatch: currently: arm-frequency-capped 
> Apr 24 17:17:16 CCTV-NVR user.notice throttlewatch: currently: arm-frequency-capped 
> Apr 24 17:17:46 CCTV-NVR user.notice throttlewatch: currently: arm-frequency-capped 
> Apr 24 17:18:16 CCTV-NVR user.notice throttlewatch: currently: arm-frequency-capped 
> Apr 24 17:18:46 CCTV-NVR user.notice throttlewatch: currently: arm-frequency-capped 
> Apr 24 17:19:17 CCTV-NVR user.notice throttlewatch: currently: arm-frequency-capped 
> Apr 24 17:19:47 CCTV-NVR user.notice throttlewatch: currently: arm-frequency-capped 
> Apr 24 17:20:17 CCTV-NVR user.notice throttlewatch: currently: arm-frequency-capped 
> Apr 24 17:20:47 CCTV-NVR user.notice throttlewatch: currently: arm-frequency-capped 
> Apr 24 17:21:05 CCTV-NVR user.err motion: [1:ml1] [ERR] [ENC] ffmpeg_put_frame: Error while writing video frame
> Apr 24 17:21:05 CCTV-NVR user.err motion: [1:ml1] [ERR] [EVT] event_ffmpeg_put: Error encoding image
> Apr 24 17:21:05 CCTV-NVR user.err motion: [1:ml1] [ERR] [ENC] ffmpeg_put_frame: Error while writing video frame
> Apr 24 17:21:05 CCTV-NVR user.err motion: [1:ml1] [ERR] [EVT] event_ffmpeg_put: Error encoding image
> Apr 24 17:21:05 CCTV-NVR user.err motion: [1:ml1] [ERR] [ENC] ffmpeg_put_frame: Error while writing video frame
> Apr 24 17:21:05 CCTV-NVR user.err motion: [1:ml1] [ERR] [EVT] event_ffmpeg_put: Error encoding image
> Apr 24 17:21:05 CCTV-NVR user.err motion: [1:ml1] [ERR] [ENC] ffmpeg_put_frame: Error while writing video frame
> Apr 24 17:21:05 CCTV-NVR user.err motion: [1:ml1] [ERR] [EVT] event_ffmpeg_put: Error encoding image
> Apr 24 17:21:05 CCTV-NVR user.err motion: [1:ml1] [ERR] [ENC] ffmpeg_put_frame: Error while writing video frame
> Apr 24 17:21:05 CCTV-NVR user.err motion: [1:ml1] [ERR] [EVT] event_ffmpeg_put: Error encoding image
> Apr 24 17:21:05 CCTV-NVR user.err motion: [1:ml1] [ERR] [ENC] ffmpeg_put_frame: Error while writing video frame
> Apr 24 17:21:05 CCTV-NVR user.err motion: [1:ml1] [ERR] [EVT] event_ffmpeg_put: Error encoding image
> Apr 24 17:21:05 CCTV-NVR user.err motion: [1:ml1] [ERR] [ENC] ffmpeg_put_frame: Error while writing video frame
> Apr 24 17:21:05 CCTV-NVR user.err motion: [1:ml1] [ERR] [EVT] event_ffmpeg_put: Error encoding image
> Apr 24 17:21:05 CCTV-NVR user.err motion: [1:ml1] [ERR] [ENC] ffmpeg_put_frame: Error while writing video frame
> Apr 24 17:21:05 CCTV-NVR user.err motion: [1:ml1] [ERR] [EVT] event_ffmpeg_put: Error encoding image
> Apr 24 17:21:05 CCTV-NVR user.err motion: [1:ml1] [ERR] [ENC] ffmpeg_put_frame: Error while writing video frame
> Apr 24 17:21:05 CCTV-NVR user.err motion: [1:ml1] [ERR] [EVT] event_ffmpeg_put: Error encoding image
> Apr 24 17:21:05 CCTV-NVR user.err motion: [1:ml1] [ERR] [ENC] ffmpeg_put_frame: Error while writing video frame
> Apr 24 17:21:05 CCTV-NVR user.err motion: [1:ml1] [ERR] [EVT] event_ffmpeg_put: Error encoding image
> Apr 24 17:21:05 CCTV-NVR user.err motion: [1:ml1] [ERR] [ENC] ffmpeg_put_frame: Error while writing video frame
> Apr 24 17:21:05 CCTV-NVR user.err motion: [1:ml1] [ERR] [EVT] event_ffmpeg_put: Error encoding image
> Apr 24 17:21:05 CCTV-NVR user.err motion: [1:ml1] [ERR] [ENC] ffmpeg_put_frame: Error while writing video frame
> Apr 24 17:21:05 CCTV-NVR user.err motion: [1:ml1] [ERR] [EVT] event_ffmpeg_put: Error encoding image
> Apr 24 17:21:06 CCTV-NVR user.err motion: [1:ml1] [ERR] [ENC] ffmpeg_put_frame: Error while writing video frame
> Apr 24 17:21:06 CCTV-NVR user.err motion: [1:ml1] [ERR] [EVT] event_ffmpeg_put: Error encoding image
> Apr 24 17:21:06 CCTV-NVR user.err motion: [1:ml1] [ERR] [ENC] ffmpeg_put_frame: Error while writing video frame
> Apr 24 17:21:06 CCTV-NVR user.err motion: [1:ml1] [ERR] [EVT] event_ffmpeg_put: Error encoding image
> Apr 24 17:21:06 CCTV-NVR user.err motion: [1:ml1] [ERR] [ENC] ffmpeg_put_frame: Error while writing video frame
> Apr 24 17:21:06 CCTV-NVR user.err motion: [1:ml1] [ERR] [EVT] event_ffmpeg_put: Error encoding image
> 

The errors above gone on until the time I rebooted the Pi.

Can someone help explain and help prevent this reoccurring
[motioneye.log](https://github.com/ccrisan/motioneye/files/1944345/motioneye.log)

Thank you!

ccrisan, 2018-04-24:
How much disk space do you have available on your external HDD?

alan-shenton, 2018-04-24:
The total disk is 1TB, with 30GB remaining. It's been that way for over 6 months, with older recordings being deleted regularly, and the file sizes being fairly consistent between 800MB and 1GB). I don't think it's down to lack of space, as the free space is always around the same amount.

ccrisan, 2018-04-24:
> user.notice throttlewatch: currently: arm-frequency-capped

That happens when the Raspberry Pi overheats, as far as I know. Could it be that the hot season is here and things are simply hotter than before?

JamBalaya56562, 2026-11-07:
Closing, as this was on motionEyeOS, the recordings came back after a reboot, and the log showed the Pi throttling (`arm-frequency-capped`), which ccrisan pointed to as overheating. motionEyeOS is maintained in its own repository (motioneye-project/motioneyeos), so please report it there if this happens again.

closed 2026-11-07 by JamBalaya56562