motioneye-project/motioneye #868: Possible to get MotionEye Loaded into BlueIris Surveillance Software

hoofta, 2018-04-24:
I have tried just about every option to get my MotionEye camera running off of a Raspberry Pi Zero loaded into BlueIris so this camera stream can sit along with my other cameras. After hours of head-pounding frustration, I have turned to the all-wise and knowing Internet, more specifically... Github!

kolonuk, 2018-05-10:
Certainly no expert, but I agree with Garret, you should be able to just use the motion daemon.  I used to run bare motion on a Pi before I found motioneye, but not with the official camera, just a bunch of networked cameras.

Have you got it working with VLC?  That would be the first step to test the stream actually works.

You should check out [this Stack Exchange question](https://raspberrypi.stackexchange.com/questions/23182/how-to-stream-video-from-raspberry-pi-camera-and-watch-it-live)

As suggested, you should ask specific questions on the blueiris forums (I'm sure someone has done it before), and post any error messages to do with the camera on the raspi StackExchange.

austwhite, 2020-06-04:
I believe, if you are referring to MotionEyeOS on the Raspberry, you should be able to just use the RTSP stream in Blue Iris.  
As mentioned above, see if you can stream it into VLC or something similar and then work from there.

JamBalaya56562, 2026-11-08:
Closing as answered above: Blue Iris can take the camera's MJPEG stream from Motion directly, without going through motionEye's web UI. With Video Streaming on, its address is under Useful URLs › Streaming URL. Check that it plays in VLC first.

closed 2026-11-08 by JamBalaya56562