motioneye-project/motioneye #870: Running on port 80 alongside apache?

dragonflyuk, 2018-04-25:
Is there anyway to have MotionEye work in a web subdirectory, so the address would be something like http://raspberrypi.local/cctv

I already have it working on port 80, but I really need it working alongside an apache install on the same port, so that both can be accessed from dataplicity?

ccrisan, 2018-04-26:
This is not a valid motionEye issue. Please look for help on Apache docs/forums.

closed 2018-04-26 by ccrisan

dragonflyuk, 2018-04-26:
Not really, how can I change where motion eye responds to, i.e cctv subdirectory instead of root?

ccrisan, 2018-04-26:
@dragonflyuk it's up to Apache to route requests coming to e.g. `/your/cams/path` to motionEye's "root". [Running Behind Nginx](https://github.com/ccrisan/motioneye/wiki/Running-Behind-Nginx) might help.

dragonflyuk, 2018-04-28:
Thanks @ccrisan that works perfectly, I can actually do all I need with nginx.

ccrisan, 2018-04-28:
@dragonflyuk I know :) Nginx kind of rules.