motioneye-project/motioneye #872: Ubuntu 18.04 LTS

ostermayer, 2018-05-02:
After upgrading from 16.04LTS --> 18.04LTS motioneye would not start as a service.
The issue seems to be a problem with libcurl3 in 18.04TLS.

This error was generated when running: "meyectl startserver -b -c /etc/motioneye/motioneye.conf"

The mysterious fix was "apt install libcurl3 python-pip"
then restart motioneye service and all was well.

chrisf4lc0n, 2018-05-10:
@ostermayer a clean installation of the 18.04 does not seem to have this problem. Following the Wiki Guide for Ubuntu goes as smooth as possible.

wandering213, 2018-06-20:
I posted this in 1508 as well

This issue is caused by old cache information left over from the old version of 
Ubuntu. Run rm -rf ~/.cache/pip and reinstall pycurl. 

https://stackoverflow.com/questions/50683895/importing-pycurl-with-libcurl4-raises-importerror

m4t7e0, 2019-07-23:
> I posted this in 1508 as well
> 
> This issue is caused by old cache information left over from the old version of
> Ubuntu. Run rm -rf ~/.cache/pip and reinstall pycurl.
> 
> https://stackoverflow.com/questions/50683895/importing-pycurl-with-libcurl4-raises-importerror
Following this  instruction i fix  this error "ImportError: /usr/lib/x86_64-linux-gnu/libcurl.so.4: version `CURL_OPENSSL_3' not found (required by /usr/local/lib/python2.7/dist-p" on debian stretch with
pip install --upgrade pycurl
! i hope will be usefull 

Thanks!

velcis, 2020-10-20:
> I posted this in 1508 as well
> 
> This issue is caused by old cache information left over from the old version of
> Ubuntu. Run rm -rf ~/.cache/pip and reinstall pycurl.
> 
> https://stackoverflow.com/questions/50683895/importing-pycurl-with-libcurl4-raises-importerror

Thanks, worked like a charm!!

JamBalaya56562, 2026-11-07:
Closing as answered: as the comments confirm, this came from a pip cache left over from the old Ubuntu version, and removing `~/.cache/pip` and reinstalling pycurl fixes it. This was with Python 2; for anyone installing now, the current steps are in the [README on `main`](https://github.com/motioneye-project/motioneye/blob/main/README.md).

closed 2026-11-07 by JamBalaya56562