motioneye-project/motioneye #878: Docker container fails to start.

vvallee, 2018-05-17:
Using extras/Dockerfile

```
vv@aelin-macbookpro:~/docker/motioneye$ docker build .
Sending build context to Docker daemon  3.072kB
Step 1/20 : FROM ubuntu:15.04
 ---> d1b55fd07600
Step 2/20 : MAINTAINER Conor Heine <conor.heine@gmail.com>
 ---> Using cache
 ---> 6e1401d19c35
Step 3/20 : ENV DEBIAN_FRONTEND noninteractive
 ---> Using cache
 ---> cdddc4d8bee7
Step 4/20 : RUN locale-gen en_US.UTF-8
 ---> Using cache
 ---> 4e0f542bfe85
Step 5/20 : ENV LANGUAGE en_US.UTF-8
 ---> Using cache
 ---> 9c5138717f02
Step 6/20 : ENV LC_ALL en_US.UTF-8
 ---> Using cache
 ---> a3738c91b1df
Step 7/20 : ENV LANG en_US.UTF-8
 ---> Using cache
 ---> 4f1bb5f206c9
Step 8/20 : ENV LC_TYPE en_US.UTF-8
 ---> Using cache
 ---> ecf67838f5c7
Step 9/20 : ENV TZ America/Los_Angeles
 ---> Using cache
 ---> e8687b25db2e
Step 10/20 : RUN apt-get update &&     apt-get --yes install     	automake 	curl 	autoconf 	build-essential 	ffmpeg 	git 	libav-tools 	libavcodec-dev 	libavformat-dev 	libavutil-dev 	libcurl4-openssl-dev 	libjpeg-dev 	libssl-dev 	libswscale-dev 	pkgconf         python-dev         python-pip 	python-setuptools 	subversion 	v4l-utils &&     apt-get clean
 ---> Running in 122b117aedfe
Ign http://archive.ubuntu.com vivid InRelease
Ign http://archive.ubuntu.com vivid-updates InRelease
Ign http://archive.ubuntu.com vivid-security InRelease
Ign http://archive.ubuntu.com vivid Release.gpg
Ign http://archive.ubuntu.com vivid-updates Release.gpg
Ign http://archive.ubuntu.com vivid-security Release.gpg
Ign http://archive.ubuntu.com vivid Release
Ign http://archive.ubuntu.com vivid-updates Release
Ign http://archive.ubuntu.com vivid-security Release
Ign http://archive.ubuntu.com vivid/main amd64 Packages/DiffIndex
Ign http://archive.ubuntu.com vivid/restricted amd64 Packages/DiffIndex
Err http://archive.ubuntu.com vivid/main Sources
  404  Not Found [IP: 91.189.88.149 80]
Err http://archive.ubuntu.com vivid/restricted Sources
  404  Not Found [IP: 91.189.88.149 80]
Err http://archive.ubuntu.com vivid/universe Sources
  404  Not Found [IP: 91.189.88.149 80]
Err http://archive.ubuntu.com vivid/universe amd64 Packages
  404  Not Found [IP: 91.189.88.149 80]
Err http://archive.ubuntu.com vivid-updates/main Sources
  404  Not Found [IP: 91.189.88.149 80]
Err http://archive.ubuntu.com vivid-updates/restricted Sources
  404  Not Found [IP: 91.189.88.149 80]
Err http://archive.ubuntu.com vivid-updates/universe Sources
  404  Not Found [IP: 91.189.88.149 80]
Err http://archive.ubuntu.com vivid-updates/main amd64 Packages
  404  Not Found [IP: 91.189.88.149 80]
Err http://archive.ubuntu.com vivid-updates/restricted amd64 Packages
  404  Not Found [IP: 91.189.88.149 80]
Err http://archive.ubuntu.com vivid-updates/universe amd64 Packages
  404  Not Found [IP: 91.189.88.149 80]
Err http://archive.ubuntu.com vivid-security/main Sources
  404  Not Found [IP: 91.189.88.149 80]
Err http://archive.ubuntu.com vivid-security/restricted Sources
  404  Not Found [IP: 91.189.88.149 80]
Err http://archive.ubuntu.com vivid-security/universe Sources
  404  Not Found [IP: 91.189.88.149 80]
Err http://archive.ubuntu.com vivid-security/main amd64 Packages
  404  Not Found [IP: 91.189.88.149 80]
Err http://archive.ubuntu.com vivid-security/restricted amd64 Packages
  404  Not Found [IP: 91.189.88.149 80]
Err http://archive.ubuntu.com vivid-security/universe amd64 Packages
  404  Not Found [IP: 91.189.88.149 80]
Err http://archive.ubuntu.com vivid/main amd64 Packages
  404  Not Found [IP: 91.189.88.149 80]
Err http://archive.ubuntu.com vivid/restricted amd64 Packages
  404  Not Found [IP: 91.189.88.149 80]
W: Failed to fetch http://archive.ubuntu.com/ubuntu/dists/vivid/main/source/Sources  404  Not Found [IP: 91.189.88.149 80]

W: Failed to fetch http://archive.ubuntu.com/ubuntu/dists/vivid/restricted/source/Sources  404  Not Found [IP: 91.189.88.149 80]

W: Failed to fetch http://archive.ubuntu.com/ubuntu/dists/vivid/universe/source/Sources  404  Not Found [IP: 91.189.88.149 80]

W: Failed to fetch http://archive.ubuntu.com/ubuntu/dists/vivid/universe/binary-amd64/Packages  404  Not Found [IP: 91.189.88.149 80]

W: Failed to fetch http://archive.ubuntu.com/ubuntu/dists/vivid-updates/main/source/Sources  404  Not Found [IP: 91.189.88.149 80]

W: Failed to fetch http://archive.ubuntu.com/ubuntu/dists/vivid-updates/restricted/source/Sources  404  Not Found [IP: 91.189.88.149 80]

W: Failed to fetch http://archive.ubuntu.com/ubuntu/dists/vivid-updates/universe/source/Sources  404  Not Found [IP: 91.189.88.149 80]

W: Failed to fetch http://archive.ubuntu.com/ubuntu/dists/vivid-updates/main/binary-amd64/Packages  404  Not Found [IP: 91.189.88.149 80]

W: Failed to fetch http://archive.ubuntu.com/ubuntu/dists/vivid-updates/restricted/binary-amd64/Packages  404  Not Found [IP: 91.189.88.149 80]

W: Failed to fetch http://archive.ubuntu.com/ubuntu/dists/vivid-updates/universe/binary-amd64/Packages  404  Not Found [IP: 91.189.88.149 80]

W: Failed to fetch http://archive.ubuntu.com/ubuntu/dists/vivid-security/main/source/Sources  404  Not Found [IP: 91.189.88.149 80]

W: Failed to fetch http://archive.ubuntu.com/ubuntu/dists/vivid-security/restricted/source/Sources  404  Not Found [IP: 91.189.88.149 80]

W: Failed to fetch http://archive.ubuntu.com/ubuntu/dists/vivid-security/universe/source/Sources  404  Not Found [IP: 91.189.88.149 80]

W: Failed to fetch http://archive.ubuntu.com/ubuntu/dists/vivid-security/main/binary-amd64/Packages  404  Not Found [IP: 91.189.88.149 80]

W: Failed to fetch http://archive.ubuntu.com/ubuntu/dists/vivid-security/restricted/binary-amd64/Packages  404  Not Found [IP: 91.189.88.149 80]

W: Failed to fetch http://archive.ubuntu.com/ubuntu/dists/vivid-security/universe/binary-amd64/Packages  404  Not Found [IP: 91.189.88.149 80]

W: Failed to fetch http://archive.ubuntu.com/ubuntu/dists/vivid/main/binary-amd64/Packages  404  Not Found [IP: 91.189.88.149 80]

W: Failed to fetch http://archive.ubuntu.com/ubuntu/dists/vivid/restricted/binary-amd64/Packages  404  Not Found [IP: 91.189.88.149 80]

E: Some index files failed to download. They have been ignored, or old ones used instead.
The command '/bin/sh -c apt-get update &&     apt-get --yes install     	automake 	curl 	autoconf 	build-essential 	ffmpeg 	git 	libav-tools 	libavcodec-dev 	libavformat-dev 	libavutil-dev 	libcurl4-openssl-dev 	libjpeg-dev 	libssl-dev 	libswscale-dev 	pkgconf         python-dev         python-pip 	python-setuptools 	subversion 	v4l-utils &&     apt-get clean' returned a non-zero code: 100
```

kleini, 2018-05-17:
Your extras/Dockerfile is totally outdated. Actual Dockerfiles base on Ubuntu 17.10 images and not any more on Ubuntu 15.04. You need to update your Dockerfiles.

kleini, 2018-05-17:
But why do you want to build the Docker images on your own? Use the ready to use images:
    docker pull ccrisan/motioneye:dev-amd64

closed 2018-05-17 by kleini

vvallee, 2018-05-17:
You were correct I had a very old version of the repo.