motioneye-project/motioneye #880: Low Res Stream for motion detection - Hi Res Stream for recording ?

ZackSc0tt, 2018-05-19:
Hey there.

I wonder if there is any possibility to set up a low resolution stream for motion to detect motion in the stream and then record the videos from the high resolution stream? My Network Cams all have at least 2 Streams. One for Hi Res 720P one for low res.

All because i want to get rid of this high cpu usage of the motion daemon.
Would be nice if a function would be implemented to do so.
Or does anybody have a workaround through the config files? 

Thanks in advance!

janpascal, 2018-06-01:
Hi, that's easy to do: if you use the latest version of motion (I believe 4.1 or newer), it supports the netcam_highres option. See the [motion documentation](https://motion-project.github.io/motion_config.html#netcam_highres).
In the camera configuration in motioneye, under the "Extra motion Options", enter an option such as

`netcam_highres rtsp://192.168.x.y:554/Streaming/Channels/101`

ZackSc0tt, 2018-06-02:
@janpascal You're awesome. I didnt noticed that feature! Many thanks ! This will very improve my setting !

zagrim, 2025-12-15:
@cheperboy I don't recall seeing that kind of option for Motion, either. However, according to [Motion docs](https://motion-project.github.io/motion_config.html#netcam_high_url) for at least the most recent version, the images saved when using netcam_high_url (that preceded netcam_highres in Motion 4.4) should anyway be from the high resolution feed:
> Motion detection on the normal resolution will trigger the saving of images from the high resolution stream.